bambulab/BambuStudio #296: Mac Pro 2013 video is blue/green lines or sometimes blinking

smolz, 2022-09-18:
Just got my printer and install the slicer on my 2013 Mac Pro. When I go to the device screen and start the video stream I get what is in the [screenshot](https://imgur.com/a/TXCPIar).  If I move the app around the screen the stream will display in quick flashes with it alternating between the lines and the video.  The video seems to also be really sped up.  My installation on a new MacBook Pro with M1 seems to be okay and the Handy app seems ok.
[image]

bambu123, 2022-09-20:
Thank you for your report. I'll find it out.

smolz, 2022-10-17:
As another data point.  https://www.reddit.com/r/BambuLab/comments/y4wvm5/camera_feed_in_studio_broken/

bambu123, 2022-10-18:
Have you tried the new version of BambuStudio?

smolz, 2022-10-18:
Assuming the latest version is 01.02.00.09, then it is the same.
[image]
[image]

smolz, 2022-11-05:
Version 1.03.00.12 works sometimes for several seconds and then either freezes or shows green

bambu123, 2022-11-10:
What's the OS version? It seems to be the problem with the hardware video decoder. Could you update your macOS?

smolz, 2022-11-12:
> What's the OS version? It seems to be the problem with the hardware video decoder. Could you update your macOS?

12.6.1 - latest Monterey.  Looks like there was a new release this week of studio (1.03.00.12).  This appears to be much better, then the previous version.  Not perfect but the video appears to be good 90+/- %.  There are now occasional flashes of green or lines, and occasionally frozen video for a second, but is for the part at least useable.

bambu123, 2023-02-06:
@smolz Does the recent 1.4.2 version has some problem?
If it does, please send some pictures or videos to show the problem.

smolz, 2023-02-06:
Attached is a quick screen recording. It overall seems better as it is not always as shown.  It comes and goes for whatever reason.
https://user-images.githubusercontent.com/3820601/216916498-9640a823-a48f-4422-bc4b-3b28676bf777.mp4

bambu123, 2023-02-06:
@smolz Thanks very much. It's helpful.

bambu123, 2024-02-06:
Resolved, please follow the prompts on Studio to update the plugin to 01.08.04.05.

closed 2024-02-06 by bambu123