bambulab/BambuStudio #297: Z-offset cannot be adjusted (for textured PEI plate).

RobinWhitfield, 2022-09-19:
Due to the nature of the textured PEI plate, the nozzle typically wants to be slightly closer to the print surface or you can end up with gaps on the first layer infill.
For some filaments (PETG) you don't however want to have such an offset or they can stick too well and damage the plate surface, so this cannot be a universally set value for the plate.

It would be great to have this as a per-filament option, or ideally two. One for general printing (default 0), and another for the textured PEI plate.

JstBecause, 2022-09-19:
I second this. It would even be nice if we had baby stepping ability as well!

I have had good luck with making my first layer .25 on the energetic PEI.

But they won't support anything unless it's theirs.

RobinWhitfield, 2022-09-19:
They've said they'll look to support other printers within Bambu Studio in the future, and their forthcoming textured PEI is likely very similar to the Energetic one if not made by them, so will benefit from the ability to set a z-offset.

SaltWei, 2022-09-20:
This seems to be odd.
Does some other slicers have same option in per-filament setting page?

RobinWhitfield, 2022-09-21:
> This seems to be odd. Does some other slicers have same option in per-filament setting page?

No, but they typically have a z-offset in the machine profile, that people use to increase/reduce adhesion when printing certain materials on textured plates.

Other slicers don't have multiple bed plate options either though.

On the X1 the only reason I think why someone may want to add a z-offset is for the textured plate, because the texture means that ABL will record the height of the "top" of the texture, and therefore may not go into the texture much, or may end up with gaps as some material does go into the texture making the line width on the surface narrower.

On the face of it this would make it seem like you'd want to be able to set a z-offset for the plate itself (in the machine profile), but because some filaments can stick too well (PETG) on textured PEI, I think it needs to be a per-filament, per-plate configuration. option.

dereknheiley, 2022-09-22:
I haven't noticed any gap printing on my energetic textured pei sheet in petg. I honestly just tossed it in there with bed leveling enabled and didn't give it a second thought.

timixretroplays, 2022-10-12:
Having received my textured PEI plate, this now seems particularly important. I can't get a single successful PLA or ABS print done with it, and others have reported similar issues.

It does appear as though it prints slightly too high off the tPEI plate to adhere properly. Discussion on Discord: https://discord.com/channels/970163353542606888/1028390423976230932

An option to alter the Z height for printing on this plate is now kind of urgent as this plate is basically useless to me without it, and it'll be affecting other users as well.

RobinWhitfield, 2022-10-12:
@SaltWei, this is not a printer issue. It's very much a slicer feature request.

SaltWei, 2022-10-12:
> @SaltWei, this is not a printer issue. It's very much a slicer feature request.

For Bambu Lab printer, the z height should not be wrong or need offset to compensate. If need to get higher bed adhension by z-offset, the the bed level function should be checked.

SaltWei, 2022-10-12:
So we mark this as printer issue, and printer guys will see it.

RobinWhitfield, 2022-10-12:
I understand that generally, but you're missing the point around textured plates and why we want this.

Surely you can understand that with a textured plate, the ABL measures the top of the texture, and that the z-offset wants to be set to approximately half of the thickness of the texture.

For PETG however if you do this it will stick too well, so you want to "float" it on the top of the texture as is being done right now.

RobinWhitfield, 2022-10-12:
As some people will always be using third party beds, there will always be a need for this setting in the slicer. Even ignoring the PETG issue it cannot be automatically set in the printer based on bed type.

VICLER, 2022-10-14:
I am using original Bambu textured plate and the first layer is just to big, so I have unfilled holes on the first layer. This is definitely a printer issue because it is wrong and can not be fixed by user now. Hope it will be fixed with the next update

pixie-wrangler-irl, 2022-11-15:
This is primarily a slicer feature that is missing pure and simple. Z offset is literally the most basic level of functionality. I honestly have no idea how this is still missing in 2022. The ability to change the first layer squish to suit the specific filament is a must have on every filament.

RobinWhitfield, 2022-11-16:
I think there's a lot of misundersanding around z-offset, which is presumably why Bambu Lab are hesistant to make it a feature.

For a perfectly smooth build plate on the X1, z-offset is *not* required unless there's a fault with bed homing (which should not be worked around by adjusting z-offset.
Layer squish is only relevant when you're using a build surface that requires you to squish the filament down into it.

If you're getting underextrusion on a smooth plate then you likely need to increase the filament flow rate.
If you're getting perfect extrusion but poor adhesion then you want to start looking at your bed surface or bed temperature, rather than overextruding that first layer with a negative z-offset.

The reason we need z-offset is for the textured plate, as underextrusion is guaranteed when running ABL that's touching off on the top of the texture. We need a z-offset of approximately half of the texture depth. It should probably be limited to no more than perhaps 50-60% of the first layer height though to try and ensure the nozzle doesn't ever end up scraping on the bed surface.

However making this a universal plate setting stops people from taking advantage of the reduced contact when printing with PETG. You probably wouldn't want a negative z-offset for printing PETG on the textured plate without a release agent as that can cause too much adhesion and damage to the plate.

suit, 2023-03-02:
this is very close related to this issue https://github.com/bambulab/BambuStudio/issues/941#issuecomment-1451729904

the PEI sheet has a z-offset of -0,04

hotellonely, 2025-02-05:
And now years later people are still facing this problem with SuperTack sometimes...

rinkek, 2026-05-09:
And here we are, one year later.
I would like to see this setting too.
But i actually want to add your own plate with their own offsets