bambulab/BambuStudio #298: Bridge lines added on top surface, when not bridging.

RobinWhitfield, 2022-09-19:
When slicing [this model](https://www.thingiverse.com/thing:36200) solidly at 0.28mm, the slicer decides that sections of the top surface should be printed as bridges.
[image]

[image]

SaltWei, 2022-09-20:
This is corner case caused by the option "Only one wall on top surfaces"

SaltWei, 2022-10-24:
This model has large overhang and the edge area is absolutely float in air. So it's considered as bridge.
Same with prusa. So maybe not a bug or just a threshold issue which is used to detect smallest bridge area?

SaltWei, 2022-11-28:
Thanks for reporting issue to us! 
Could you share this model for us to test the fix?

RobinWhitfield, 2022-11-28:
The model mentioned originally was this one: https://www.thingiverse.com/thing:36200

YifanWuBambu, 2022-12-16:
This is fixed in v1.4.