bambulab/BambuStudio #299: [Feature Request] Thermostatic Control of Exhaust Fan

ptyork, 2022-09-20:
Currently, the only control for the exhaust fan is hidden behind the start/stop G-code for each filament. A far more logical (and quiet) solution would be to have a target chamber temperature for each filament. Ramp up the exhaust fan **only** as the chamber temp approaches this threshold.

steezyprintz, 2022-09-22:
I would like this to be a feature

CNCXPROUK, 2022-10-01:
Me too.  That exhaust fan is noisy!  It just goes full speed and nothing else!

steezyprintz, 2022-10-02:
Perhaps the new chamber temp setting in the latest update now controls this.

[image]

CNCXPROUK, 2022-10-02:
Possibly yeah. I’ve just looked at Bambu handy and that would make sense.  You can’t change it like you can the other temperatures on the fly, hot end etc.

CNCXPROUK, 2022-10-02:
My printer is waiting for new filament to arrive in the post to finish a print. Currently the fan is going full speed whilst I wait

ptyork, 2022-10-02:
> Perhaps the new chamber temp setting in the latest update now controls this.

Is that a nightly build or something? Haven't seen that in the latest Windows build. Regardless, yes, something like that seems exactly what I'm asking for. Does the advanced tab still have start/end g-code that sets the fan speed (M106)? If so, then this UI change might just be getting ready for this more advanced chamber fan control. Perhaps waiting on the next firmware revision.

Won't know without an official Bambu response here, though. Anything you can confirm or deny @YifanWuBambu?