bambulab/BambuStudio #302: HMS tab displays blank warnings

labatt, 2022-09-20:
In slicer 1.02.00.09, when you go to Device > HMS, the badge is shown but the text is not.

[image]

StoneLiBambulab, 2022-09-21:
Hi labatt,
       Thanks for your feedback. And could you send us the Bambu Studio logs so we can check if there's something wrong there. The files can be accessed on this location:
C:\Users\user.name\AppData\Roaming\BambuStudio\log

BambuLab Studio Team

OmiCron07, 2022-10-20:
The release notes say that the bug is fixed, but is still there in the latest version:

[image]

closed 2024-01-15 by SaltWei