bambulab/BambuStudio #303: [Enhancement] Ability to choose the layer number to view in the Preview pane

dnhkng, 2022-09-20:
I need to check the first layer of a print in the preview window, as sometimes I spot slicer errors. However, on tall prints, it's very hard to get to layer 1, or any particular layer, and even tiny mouse movements make the slider jump several layers at a time.

There is already a layer number box attached to the slider, it would be great if we could double-click it, and manually type the layer we want to jump to.

YifanWuBambu, 2022-09-21:
@dnhkng 
“even tiny mouse movements make the slider jump several layers at a time”
This is a bug and we will get it fixed.

You may go to upper/lower layer accurately by pressing "up/down" arrow key on the keyboard for now.

SaltWei, 2023-03-02:
Already had

closed 2023-03-02 by SaltWei