bambulab/BambuStudio #306: Feature request-Pause if door is open

coat64, 2022-09-20:
Besides enjoying this printer, I am testing it for use at schools. Then it is all about safety. A pause when the door opens would be mandatory. For all other users this would be very annoying :). So adding it as an option would be great.

ptyork, 2022-09-21:
Yep it would be annoying. :) Seriously, though, I don't think the hardware exists to support this. No sensor on the door and the camera is facing the wrong way to use computer vision. Probably would need to have a door sensor/switch added to X1v2. Until then, perhaps someone could design a door latch+lock of some sort...which might be even more annoying.

RobinWhitfield, 2022-09-21:
> No sensor on the door

That's where you're wrong. There's a small microswitch hidden under the rubber on the right hand side.

I'd like to see this as an optional safety feature, but for personal use it would be very annoying.

Some prints are better to be done with the door open too.

ptyork, 2022-09-23:
> > No sensor on the door
> 
> That's where you're wrong. There's a small microswitch hidden under the rubber on the right hand side - it's just not currently configured to do anything.

I stand corrected. I saw a YouTube video saying it lacked a door switch and never actually looked for myself. Cute little bugger...

RobinWhitfield, 2022-09-23:
> I saw a YouTube video saying it lacked a door switch and never actually looked for myself.

People put all sorts of drivel on the internet 😁

psiberfunk, 2022-09-24:
As a dirty hack you could put a smart door sensor on the door and cut the power hard as a failsafe .. would ruin prints but guarantee safety …