bambulab/BambuStudio #307: Price in Filament settings for Support W

rwoldberg, 2022-09-20:
The default price for Support W Filament in the filament settings is 29.99/kg it should be 69.98/kg for the US.  Maybe this setting should be allowed to be updated for the default filaments rather than creating a copy.  This is a minor issue but it does change the total cost on the total estimation screen after slicing.

RobinWhitfield, 2022-09-23:
This is definitely a complication for the default profiles, as it's always going to vary based on currency.

It's certainly not <currency>69.98/kg everywhere, although neither is it <currency>29.99/kg.

Personally I think I'd be inclined to remove the pricing for the default profiles. Better it says nothing than gives an incorrect figure.

rwoldberg, 2022-09-23:
Well, maybe there could be a table that is separate from the default profiles that it pulls the cost from.  This cost could even be pulled from the web and updated from the Bamboo store based on the country you have selected, that way it would always be up to date.

SaltWei, 2022-10-12:
Good idea, but now we only support manual updating in default profile.
Auto updating from web is feature request and not very important or urgent.

YifanWuBambu, 2022-10-21:
Fixed in v.1.3.0.12.

closed 2022-10-21 by YifanWuBambu