bambulab/BambuStudio #308: feature - calibrate before print

coat64, 2022-09-20:
Maybe a terrible idea, but calibrating before a print costs (subjectively) a long time. Maybe it is possible that one can get the printer already calibrated (bed-leveling and/or flow calibration) before you slice your model? Just like the preheating on old-fashioned printers. Slicing normally takes a few minutes by checking and tweaking. It would be great if the printer is then ready to print the model.

YifanWuBambu, 2022-09-21:
@coat64 
The printer will remember last calibration result.
So, if you use the same filament, please skip it.

For bed-leveling, it is suggested to do it everytime. We will consider whether to add an option of performing bed leveling before print job.

RobinWhitfield, 2022-09-21:
> Maybe it is possible that one can get the printer already calibrated (bed-leveling and/or flow calibration) before you slice your model?

The issue with this is that you really want the bed to be up to temperature before you to ABL, as it will expand and change shape slightly at higher bed temperatures. The printer doesn't know the bed temperature for your print until you've sliced the model and sent it over.

coat64, 2022-09-21:
I can see that, it is more a (i think handy) convenience which you have to start yourself at the right time. Selecting the profile and filament is probably the first thing you do. After that do the bed leveling (and the flow calibration if needed) before getting into the details of the slicing-settings, inspecting the preview etc. If you did change the temps, select bed-leveling again if you want. Downside could be that if you are quick you have to wait to send the print until it is ready.

parksddd, 2022-09-21:
I like this feature idea, but at the very least can the slicer ask the printer what the previous filament was, and if it is the same prompt the user when sending the print to the printer to say "This is the same filament, you can skip the flow calibration". It is a small change but it would certainly be welcome especially for non-AMS users.

YifanWuBambu, 2022-09-22:
@parksddd 
An easy way is toggle off “flow calibration” in the “Send print job” dialog if you confirm that the filament does not changed.

coat64, 2022-09-22:
Something like this, it doesn't have to be at the front.
[image]