bambulab/BambuStudio #309: Camera locked up

ken-kaplan, 2022-09-21:
After some random period of time Bambu Studio (and the Handy app) cannot connect to the camera. Studio reports Failed [0] error. Handy pauses for a moment and then nothing. I've restarted the machine, rebooted my router and no change. The camera begins working at unpredictable times.

Studio is running on a MacBook with the latest software. Handy on an Android, also latest software.

bambu123, 2022-09-26:
For the studio, please send the log file to us. You can locate log files from the menu "Help" > "Show Configuration Folder" and logs are in the subdirectory "log"。

fryertruck, 2022-09-28:
Happens to me every day. Working fine again after rebooting the printer.

```
2022-09-28 16:04:11.160072[Thread 0x000000011dea9600]:MediaPlayCtrl::SetStatus: Laden...
2022-09-28 16:04:11.619029[Thread 0x000000011dea9600]:light: true
2022-09-28 16:04:11.940040[Thread 0x000000011dea9600]:[Bambu_Open] call IOTC_Connect_ByUIDEx(WHR83B3G2DK2N965111A).......
2022-09-28 16:04:12.260539[Thread 0x00007000016aa000]:MediaPlayCtrl: end load
2022-09-28 16:04:12.260539[Thread 0x000000011dea9600]:[Bambu_Open] IOTC_Connect_ByUIDEx failed SID[-90]
2022-09-28 16:04:12.260574[Thread 0x00007000016aa000]:MediaPlayCtrl: start load
2022-09-28 16:04:12.260696[Thread 0x000000011dea9600]:MediaPlayCtrl::onStateChanged: 0, last_state: 5
2022-09-28 16:04:12.260707[Thread 0x000000011dea9600]:MediaPlayCtrl::onStateChanged: skip when task not finished
2022-09-28 16:04:12.260711[Thread 0x000000011dea9600]:MediaPlayCtrl::onStateChanged: 0, last_state: 5
2022-09-28 16:04:12.260714[Thread 0x000000011dea9600]:MediaPlayCtrl::onStateChanged: skip when task not finished
2022-09-28 16:04:12.260717[Thread 0x000000011dea9600]:MediaPlayCtrl::onStateChanged: 0, last_state: 5
2022-09-28 16:04:12.260720[Thread 0x000000011dea9600]:MediaPlayCtrl::onStateChanged: skip when task not finished
2022-09-28 16:04:12.261004[Thread 0x000000011dea9600]:[Bambu_Open] call IOTC_Connect_ByUIDEx(WHR83B3G2DK2N965111A).......
2022-09-28 16:04:12.578959[Thread 0x000000011dea9600]:[Bambu_Open] IOTC_Connect_ByUIDEx failed SID[-90]
```

```
2022-09-28 16:04:24.612815[Thread 0x00007000016aa000]:MediaPlayCtrl: end load
2022-09-28 16:04:24.612834[Thread 0x000000011dea9600]:[Bambu_Open] IOTC_Connect_ByUIDEx failed SID[-90]
2022-09-28 16:04:24.612998[Thread 0x000000011dea9600]:MediaPlayCtrl::onStateChanged: 0, last_state: 5
2022-09-28 16:04:24.613013[Thread 0x000000011dea9600]:MediaPlayCtrl::onStateChanged: size: 0x0
2022-09-28 16:04:24.613117[Thread 0x000000011dea9600]:MediaPlayCtrl::SetStatus: Gestoppt.
2022-09-28 16:04:24.613390[Thread 0x000000011dea9600]:MediaPlayCtrl::SetStatus: Laden fehlgeschlagen [0]!
2022-09-28 16:04:24.613524[Thread 0x000000011dea9600]:MediaPlayCtrl::onStateChanged: 0, last_state: 3
2022-09-28 16:04:24.613535[Thread 0x000000011dea9600]:MediaPlayCtrl::onStateChanged: 0, last_state: 3
```

cmguo, 2022-09-29:
Yes, the error code shows the printer is not online. If it would happen again, please report it with the mobile app "Bambo Handy".
* Install Bambo Handy from here: "https://bambulab.com/en".
* In Personal > Support Tickets, press the '+' button.
* Input time range contains the error live view.

fryertruck, 2022-10-17:
I think the problem occurs only when the router loses the Internet connection in the meantime. When the router reconnects, the printer does not manage to reconnect to the Bambu server.

ken-kaplan, 2022-10-17:
It's not the router losing connection to the internet, it's the printer losing the connection. My router logs show continuous connectivity to the internet. It's only the camera that is effected  - all other information from the printer is served in real time. Either the camera feed is timing out and cannot re-establish it's connection or the camera is the only resource using IoT protocol and it can only reconnect with a full restart of the printer. My assumption is the code the makes the camera connection runs once as part of the printer start-up and anything the breaks that connection cannot be resolved without the printer restarting. It needs to catch the IoT error and attempt to reconnect rather than just showing the error. Telemetry must be served locally cause that always works whether the camera is connected or not. If the camera was served locally the problem wouldn't exist.