bambulab/BambuStudio #311: Add acceleration control for all line types instead of the current one global

steezyprintz, 2022-09-22:
Currently for example to print one of my products in black PLA+ I need to turn the acceleration of the whole print down to 4k acceleration to make it look OK, this increases the overall print time alot :(
I am use to the control of Cura of changing the acceleration of the outer and inner wall to be low for good quality and outer looks but having the infill far higher reducing overall print time. With the epic speed and acceleration of the X1 this will be a big time saving for me and many others not that happy with the outer ringing / ghosting dark prints have.

YifanWuBambu, 2022-09-23:
@steezyprintz 
This is in our plan and is related to the printer firmware.
It will be included in the recent release.

steezyprintz, 2022-09-23:
great thanks cant wait, any idea when it is due to be released? @YifanWuBambu

SoftFever, 2022-09-24:
@steezyprintz 
I have implemented this feature in my fork.
If you are interested, you can give it a try. [LINK](https://github.com/SoftFever/BambuStudio-SoftFever/releases/tag/v1.2.4-sf)
[image]

I have tested it with several prints by myself already.
It would be great if I can get more confirmations from other users before I create a PR to the official stream.

-SoftFever

YifanWuBambu, 2022-09-24:
> great thanks cant wait, any idea when it is due to be released? @YifanWuBambu

Will be in the next release, which is scheduled in late October.

CS-Account, 2022-11-21:
Was this ever added? I cant seem to find it in the latest update

closed 2022-12-16 by YifanWuBambu