bambulab/BambuStudio #312: [Feature Request] Add infill walls option to speed up print time of multi wall prints

steezyprintz, 2022-09-22:


YifanWuBambu, 2022-09-23:
Hi @steezyprintz, 
By "infill walls", do you mean inner walls?

steezyprintz, 2022-09-23:
I mean walls that are inside the inner walls, see image from cura the orange walls are infill walls.

The reason I think this is very useful for accurate printing is it reduces build up of material where it should not be, mainly corners. I mainly print with PLA+ PETG and other harder to print materials.
I tested my X1 and if you print a cube with 5 walls inside to outside, the corners will be bad but if 2 walls its good.
Better with 5 walls printed outside to inside but then bad z seam.
Being able to print 1 outer 1 inner wall and then 3 infill walls achieves stronger parts faster from my experiences with good corners/detail/accurate.

[image]

YifanWuBambu, 2022-09-24:
@steezyprintz 
If possible, could you save the model along with its settings as a 3mf (in Cura) and share it to me?
Thanks!