bambulab/BambuStudio #313: [Feature Request] Slice keyboard shortcut key and toggle auto slice

steezyprintz, 2022-09-22:
Like with some of the other slicer functions having a keyboard shortcut key such as arrange has A. Slice keyboard shortcut key = S, would great for making quick changes to see impact on print time. Also a toggle auto slice after making one change would be a nice additional for those that like to shave off time by tweaking settings like me.

YifanWuBambu, 2022-09-23:
@steezyprintz 
Bambu Studio also supports shortcut key "A" (capital) for arrangement.
[image]

Shotcut key "S" has been used for scaling.
We may find another key for slicing.

steezyprintz, 2022-09-23:
perhaps ctrl + S