bambulab/BambuStudio #314: [Feature Request] XY Hole and Contour Compensation move to Filament Settings

steezyprintz, 2022-09-22:
XY Hole and Contour Compensation move to Filament Settings, as this varies per material it makes little sense why these settings are currently in the Profile Quality section. Currently I need to have a filament settings tuned per material then need to then also create a profile just for the material again in order to get actuate dimension prints.

YifanWuBambu, 2022-09-23:
@steezyprintz 
We suggest to adjust the flow for each filament for this target.
[image]

RobinWhitfield, 2022-09-23:
@YifanWuBambu, that value does a totally different thing.

The compensation values being talked about are typically used to account for overall part shrinkage, whereas the flow ratio can only account for variations required in extrusion width.
The X-Y hole and contour compensation values are only typically required if a) the machine you're printing from isn't dialled in correctly (say stepper pulses per mm is wrong), or b) the filament you're printing with has a higher shrinkage rate.

I agree with @steezyprintz that these should be filament specific values. Perhaps don't move them though, rather add overrrides for filament, as someone may wish to slightly increase or decrease the size of a printed model for other reasons.

steezyprintz, 2022-09-23:
Thanks Robin for explaining those points. 

I somewhat agree with adding over rides however personally I would like the compensation moved to the filament settings but yes over rides could be useful to some people somewhere else when they are having parts fitting together issues. Perhaps a over ride where they are currently.

ctadlock, 2022-12-23:
X-Y countour and hole compensation should definitley be moved to a fillament setting. Every fillamant has a different value that can only be determined empirically.

Sage-Kreiter, 2022-12-23:
Definitely need this implemented. Any update on when/if this will be added??

mark-schuler, 2022-12-23:
> X-Y countour and hole compensation should definitley be moved to a fillament setting. Every fillamant has a different value that can only be determined empirically.

Some filament types can be wildly different.  Definitely, a useful tool to have.

EricCraigen, 2022-12-24:
This would be a perfect feature for many of my projects. When could we expect to see this implemented?

stony2k, 2024-02-03:
I agree. My X1 prints PLA with exact 0.2 mm undersize. When scaling the part i. X and Y to 100.12 % the real part ist about 99,99% to the real measures, which it good enough. Some shrinkage option is much appreciated instead of scaling every part every time.