bambulab/BambuStudio #316: Feature Request: Check box to exclude initial purge

kuljaden, 2022-09-23:
Hello, 

This is a feature request to add a checkbox to exclude the initial purge. That purge is unnecessary if you are printing with the same material as you did your last print and not using the AMS, so no initial purge is needed.

Thank you for all you do Bambu Labs!

YifanWuBambu, 2022-09-24:
@kuljaden,
We have add this to the plan list. Thanks for the suggestion.

YifanWuBambu, 2022-09-26:
Hi @rzucco ,
I think @kuljaden is mention about this purge.
[image]

Skipping this purge has a problem that the old filanet residues may stay at the nozzle and increase the risk of nozzle clog.
For example, we first print ABS and then print PLA. The ABS residues may not be well cleaned at PLA's print temperature 230℃.

kuljaden, 2022-09-26:
@rzucco what I am referring too is the following situation. Say I am printing with yellow PLA, i print my part, and I am all done. I want to print another part in the exact same yellow PLA. There is no difference in color/material/anything. It is not necessary to purge any filament in that case, which is why I suggested a checkbox to allow that option so the user can decide if the excess purge is necessary.

rwoldberg, 2022-09-28:
I also created the same request: https://github.com/bambulab/BambuStudio/issues/223