bambulab/BambuStudio #317: Erroneous Internal Solid Infill Made by Bambu Studio Sliced file

VACIndustries, 2022-09-23:
[image]

When slicing this file it adds a bunch of internal solid infill which should not be (see picture)

Attached is both the base.3mf file which was downloaded from here https://www.printables.com/model/239774-modular-desktop-fountain/files

And the Bambu Studio sliced file in the image above (hex_plate_1.gcode.3mf)
[3mffiles.zip](https://github.com/bambulab/BambuStudio/files/9637533/3mffiles.zip)

SaltWei, 2022-10-13:
Patch has been merged. Please wait the the next release. Will come soon.

YifanWuBambu, 2022-10-21:
Fixed in v.1.3.0.12.

closed 2022-10-21 by YifanWuBambu

Bpositivedotdk, 2022-10-30:
I think it still does this in the latest version. - Adding quite a lot of print time....
[image]

SaltWei, 2022-10-30:
> I think it still does this in the latest version. - Adding quite a lot of print time.... [image]

Please share the model for analyse

Bpositivedotdk, 2022-10-30:
> > I think it still does this in the latest version. - Adding quite a lot of print time.... [image]
> 
> Please share the model for analyse

Can I send it to you directly? don't want to share this particular model here.

SaltWei, 2022-10-30:
Yes. My email is salt.wei@bambulab.com
Thank you very much for response!

SaltWei, 2022-10-31:
Hi @Bpositivedotdk . Thanks for reporting issue.
But this is absolutely differently issue you reported. The extrusion line is not mess with floating line, but still will organized.

For the solid infill you reported, prusa has same behaviour.
[image]
Solid indill is added in slope to ensure thickness. And prusa has option "ensure vertical shell thickness" for user to disable but bambu always enable it when slicing.

So for your requirement and case, we will provided option "ensure vertical shell thickness" so you can disable these solid infill.

Bpositivedotdk, 2022-10-31:
Thanks SaltWei. Yes I see its not the floating line issue, but still a problem with solid infill. 
It would be great to have an option to remove that, so looking forward to that.
... (and in general the option to customize a lot more). Coming from Cura that has a million settings you can tweak, I find Bambu studio very limiting.)