bambulab/BambuStudio #318: Linux Version

whiiiskyy, 2022-09-24:
When is the slicer Linux version gonna be released??

The AppImage like Cura and Prusa provide works more than well. No everyone uses Windows or Mac.

Thank you

lanewei120, 2022-09-27:
> When is the slicer Linux version gonna be released??
> 
> The AppImage like Cura and Prusa provide works more than well. No everyone uses Windows or Mac.
> 
> Thank you

we are working on linux version
as there are too many issues
we can not give a stable version in a short time

however, we will submit some patches which can let Bambu Studio work on linux firstly as soon as possible

whiiiskyy, 2022-09-27:
@lanewei120 is totally understandable since you guys might have bigger problems to solve.

It's good to know tho Linux version is around the corner. Having to use Windows PC to use the slicer is not good haha

Thank for the reply and solid work the team has been done so far.

lanewei120, 2022-09-29:
just submited the codes to supprt linux version

currently some known issues:
- a random crash when software launch(we can ignore it and retry until success)
- the liveview is not working

please refer to https://github.com/bambulab/BambuStudio/issues/12 for the details

whiiiskyy, 2022-09-29:
Thank you @lanewei120 🙌

deftdawg, 2022-10-02:
@whiiiskyy this should be closed as it's a dupe of #12

whiiiskyy, 2022-10-02:
> @whiiiskyy this should be closed as it's a dupe of #12 

Sure thing. Thank you

deftdawg, 2022-10-02:
You as the author of the issue can close it...🙂
(I'm not an admin, so it might end up sitting open until the admins do a clean up)

closed 2022-11-10 by H4kunaM4tata

meladori29, 2024-10-18:
Any updates?  I got a Bambu printer and can't use it on any of my machines except the Handy app in my phone because the install on Linux has network issues. Bambu I would expect better 2 years later what is going on?

lanewei120, 2024-10-19:
@meladori29 

we have supported linux versions on github long time ago
please find it from https://github.com/bambulab/BambuStudio/releases

thanks

meladori29, 2024-10-19:
> @meladori29
> 
> we have supported linux versions on github long time ago please find it from https://github.com/bambulab/BambuStudio/releases
> 
> thanks

Yeah but the network file The studio asks you to download when you first install it says it installs but then it doesn't work you can't log into the bamboo studio and you can't send sliced files to the printer That's what I'm asking you to correct.