bambulab/BambuStudio #320: Feature Request: filament change at layer height for non-ams

CruxMason, 2022-09-24:
In addition to the pause at layer add a color change where the machine will complete an unload and then wait for filament again and do the purge and poop before starting.  Prusa slicer and the mk3s does this well.

YifanWuBambu, 2022-09-24:
Hi @CruxMason ,
Do you mean this feature?
[image]

CruxMason, 2022-09-24:
Yes

whiiiskyy, 2022-09-25:
Yes. This will also require firmware change.

IIRC, UncleJessy on YouTube showed the unload filament is gray out.
Apparently at this stage, only with AMS you can change colors.

This feature should be included not just because of color change but many of us pause the printer to insert nuts/magnets before resuming it so "Pause at layer X" is a must have feature too.

kiliankoe, 2022-09-25:
This specific feature also does not work *with* the AMS. It's completely ignored by the printer. The only way to currently get a filament change is by using the painting features. Support for this layer-based approach would be much appreciated.

[image]

YifanWuBambu, 2022-09-26:
> Yes. This will also require firmware change.
>
> IIRC, UncleJessy on YouTube showed the unload filament is gray out. Apparently at this stage, only with AMS you can change colors.
>
> This feature should [...]

[image]

YifanWuBambu, 2022-09-26:
> This specific feature also does not work _with_ the AMS. It's completely ignored by the printer. The only way to currently get a filament change is by using the painting features. Support for this layer-based approach would be much appreciated.
> 
> [image]

We will check this issue. Thanks for reporting that.