bambulab/BambuStudio #322: [Feature Request] Upload of GCode to SD-Card

mrcg, 2022-09-24:
Currently it's not possible to upload GCode to SD card.
For people who want to use other slicer this is very limiting issue in flexibility.
Also when generating the GCode with Bambu Studio i would like to have the option to select "upload to SD card".
When starting the print from SC card it's possible to select the AMC slot which is not possible if uploaded to internal storage.

One option could be to allow uploading GCode via Bambu Studio to SD card.
Alternative (even better) would be to include option in printer FW which allows to enable share of SD card in local internet via network share (e.g. samba, ftp, http). 

Please add the option to allow upload to SC card ... for GCode created with Bambu Studio but also with other alternative slicer.

whiiiskyy, 2022-09-25:
"Bambu Studio i would like to have the option to select "upload to SD card"

This is a must have feature, the most common slicers like Cura and PrusaSlicers have by default a dedicated `Export g-code` button after you have sliced the file, but again, X1 printers need to be able to read from it while providing all the features like if you were printing over Wif-Fi.

RobinWhitfield, 2022-09-25:
Just to make sure everything's clear here, you can curently export a sliced file from Bambu Studio to put on an SD card (as a .3mf, but the .gcode is contained within and can be extracted if you wanted just that), and you can place gcode from other slicers on an SD card to be installed in the printer.

Sending to the printer from Bambu Studio to have the file stored on the SD card for later printing is covered by #115 and #129 

It would be nice to be able to have direct access to the machine's FTP server to enable upload functionality to be built into other slicers though.

mrcg, 2022-09-25:
I want to send any gcode (3mf) created with any slicer from my local PC via local intranet to BBL to be stored on the SD card for later printing. The upload should be possible directly to the printer without the need to be connected to the cloud via internet.

Good would be also to have webfront on the X1 similar to Octoprint or Repetier Server to upload/download gcode and timelapse videos and also control the printer. The X1 is such a great designed product but really lacks such basic features. I hope such feature will come with later FW updates.

YifanWuBambu, 2022-09-26:
> 

We will firstly add the feature of uploading gcode (created by Bambu Studio) to the sdcard.

whiiiskyy, 2022-09-26:
> > 
> 
> We will firstly add the feature of uploading gcode (created by Bambu Studio) to the sdcard.

That is more than enough and just like how other slicers work.
It isn't recommended to download g-code from the internet anyway.

The printer however, has to accept any g-code, not just the ones generated by Bambu Studio. 
I don't have the printer yet so I can't tell if that is already possible.

YifanWuBambu, 2022-09-26:
@whiiiskyy 
>> It isn't recommended to download g-code from the internet anyway.
The g-code will be transfered to the printer's sdcard via local network (LAN).

whiiiskyy, 2022-09-26:
> The g-code will be transfered to the printer's sdcard via local network (LAN).

@YifanWuBambu that is weird. So if I don't have my printer set to use network for security reasons ( blocking data collecting, personal data collection ), how am I supposed to use the printer then???

I just installed the slicer and it only exports the file as 3mf
That is a weird behavior.

If I had the printer right now, I would need to use PrusaSlicer with X1 g-codes, slice, export g-code to the SDCard and hope it will works because Bambu Studio doesn't export g-code (???)
Or set the network by force otherwise I cannot print, I mean...

Does the printer reads and execute jobs in .3mf????

YifanWuBambu, 2022-09-26:
For G-code, I mean .gcode.3mf file, which also contains .gcode, thumbnail, slicing information, etc.
If you only want the .gcode file, please export by File=>Export=>Export G-gcode.

The X1/X1C is able to parse both .gcode.3mf and .gocde files.

whiiiskyy, 2022-09-26:
@YifanWuBambu thank you, I got the same response from Discord.
Until now we were used to .gcode file.
With X1 .gcode.3mf is all you need instead :) 

Thanks

hisptoot, 2022-10-10:
Bambulab x1/x1c have a open ftp 21 port.  You can connect to the ftp server by username: bblp and passwd: [lan_mode_access_code]. As long as you know the lan mode access code, you can use ftp at cloud mode. But the speed of current ftp server is not stable.

mrcg, 2022-10-16:
> Bambulab x1/x1c have a open ftp 21 port. You can connect to the ftp server by username: bblp and passwd: [lan_mode_access_code]. As long as you know the lan mode access code, you can use ftp at cloud mode. But the speed of current ftp server is not stable.

Thanks for this hint. FTP connection works quite good. Really very useful information!
Is it also possible to access the webcam stream via LAN only mode?

whiiiskyy, 2022-10-17:
> Is it also possible to access the webcam stream via LAN only mode?

It would be awesome to be able to access it via RTSP protocol.

hisptoot, 2022-10-17:
> Thanks for this hint. FTP connection works quite good. Really very useful information! Is it also possible to access the webcam stream via LAN only mode?

The webcam uses tutk iotc sdk. The protocol does not seem to be rtsp. Maybe use tutk iotc sdk or call some function in bambusource dll/so, then we can get video stream in lan only mode.

hisptoot, 2022-10-18:
> Thanks for this hint. FTP connection works quite good. Really very useful information! Is it also possible to access the webcam stream via LAN only mode?

Did some analysis of the new version BambuSource lib yesterday. It seems to support local lan tunnel for video streaming. But I think the firmware of the printer currently may not support this feature. I couldn't get local tunnel working.

Deadbeef64, 2023-01-17:
Where can I find this lan_mode_access_code?

H4kunaM4tata, 2023-01-17:
> Where can I find this lan_mode_access_code?

- username: bblp
- password: Printer LCD network tab, enable LAN mode and a password will be displayed
- authentication: TLS seems to be required now

Deadbeef64, 2023-01-17:
Thank you very much for this clear answer.

mrcg, 2023-01-17:
FRQ will be closed now because feature meanwhile implemented.
Thanks to Bambulab team for implementation.

closed 2023-01-17 by mrcg

RyanEwen, 2023-01-21:
Does local FTP work for the P1P or just the X series?

Deadbeef64, 2023-01-21:
I don't know. I just own an X1 and there it works (if you enable the Local LAN  option in the display menu.

RyanEwen, 2023-03-02:
Looks like FTP works on the P1P now. 

More specifically FTPS (port 990 / implicit FTP over TLS). Username bblb. Password is the access code found on the LCD in the WiFi settings. Does not require LAN-only mode.

milapointe, 2023-03-04:
> Looks like FTP works on the P1P now.
> 
> More specifically FTPS (port 990 / implicit FTP over TLS). Username bblb. Password is the access code found on the LCD in the WiFi settings. Does not require LAN-only mode.

THAT is the correct anwser even for X1. Thank you !

Macmagoo, 2023-03-07:
FTP works very well transferred some Gcode files no problems.

tschoene, 2023-03-18:
Hi, my X1C communicates over implicit FTP over TLS, but does request a client certificate!? Does anybody know, from where I can get this certificate?

milapointe, 2023-03-18:
did you tried the right port? it just asked to accept the certificate,
should not ask a certificate to be able to connect but rather a user pass
as explain in this post.

tschoene, 2023-03-18:
I tried port 990. Must I activate LAN or is it possible over wifi too? The Passcode is changing everytime when I refresh it on the display!?

milapointe, 2023-03-18:
Yup, over wifi it works and passcode change yes

Benderrodrigu3z, 2023-04-08:
> Yup, over wifi it works and passcode change yes Le sam. 18 mars 2023, 12 h 37, tschoene ***@***.***> a écrit :
> […](#)

please explain how you accomplished this? What program did you use? Mac or Windows? how did you connect to the ip? did it ask for a port if so what is it? what all info do we need for this new process?

RyanEwen, 2023-04-08:
> please explain how you accomplished this? What program did you use? Mac or Windows? how did you connect to the ip? did it ask for a port if so what is it? what all info do we need for this new [...]

https://forum.bambulab.com/t/we-can-now-connect-to-ftp-on-the-p1p/6464

Steveveepee, 2023-06-07:
For others having access denied problems with the info above. Try username: bblp  (not bblb)

markaudacity, 2024-06-24:
> Bambulab x1/x1c have a open ftp 21 port. 
I don't seem to be able to connect to it, does it have to be enabled through the front panel?

RobinWhitfield, 2024-06-24:
> > Bambulab x1/x1c have a open ftp 21 port.
> > I don't seem to be able to connect to it, does it have to be enabled through the front panel?
You now must use FTPS (note, not SFTP)