bambulab/BambuStudio #324: [Feature] Advanced nesting on buildplate

hutje, 2022-09-24:
It would be much appreciated to have a nesting feature that can calculate parts to be placed (partly) inside the surface of the other part, like a gap or hole.

Nesting is a really old tool, but a lot of companies make way too much money on this simple and basic feature.
With the current material and supply shortages, it would be really beneficial to have a opensource way to have less residual material.
Revive software like https://svgnest.com and http://deepnest.io and give it a worthy place in the new CNC/3D-printing community ✌🏻

tangketan, 2022-09-26:
Good suggestion! Actually we are working on it.

hutje, 2023-03-16:
The CNC laser software 'LightBurn' makes use of the open sourced website SVGnest. In most cases BambuStudio would be able to use it.
This could be achieved by generating a top-view contour of each part and let the software determine if the objects are able to fit together. The user would be able to see the top-view iterations and decide if the iterations are good enough.
[image]