bambulab/BambuStudio #325: {Feature Request] Add to Supports: For Support Enforcers Only

BryanHaven, 2022-09-25:
Please add the "For Support Enforcers Only" option when painting supports. Sometimes you want to add supports only to a specific overhang - PS 2.5.0 does this very well without having to paint support blockers everywhere. 

I didn't see an open issue for support blockers not being respected but may have missed it. I'll look again and open a bug if I can't find it.

YifanWuBambu, 2022-09-26:
@BryanHaven 
These options are for  your purpose.
[image]

BryanHaven, 2022-09-26:
Ok, 

 

Tried these and got various results – 

with tree supports I got what appears to be a good set of supports – 

with normal I got the same as before with the lower holes having supports in them. 

 

Thank you for looking into this

 

 <https://user-images.githubusercontent.com/110661856/192198939-f57fd7cc-0eea-42dc-88d3-afb7db0ad055.png>

YifanWuBambu, 2022-09-27:
Normal support also works well on my test model.
[image]
[image]

Would you mind sharing your model (3mf) with me?

BryanHaven, 2022-09-27:
Is it not in the projects I shared earlier? Will do when I get back to my computer.

BryanHaven, 2022-09-27:
Good morning,

 

You will need to split the model and remove the X brace – it doesn’t print well and fuses to the model – 

 

Thanks, 

 

Bryan

 

 <https://user-images.githubusercontent.com/110661856/192420567-7220b8e5-2175-4ed4-937a-17a134745998.png> 
 <https://user-images.githubusercontent.com/110661856/192420712-4e7aa912-1d94-44a1-9c7a-abfc5bbb0025.png>

YifanWuBambu, 2022-10-09:
Seems not found the model file.