bambulab/BambuStudio #326: Support Blockers not respected when slicing

BryanHaven, 2022-09-25:
Support blockers are not always respected when slicing models

[image]

[image]

SaltWei, 2022-09-26:
Could you please share the 3MF for us to reproduce?

BryanHaven, 2022-09-26:
[Base_plate_1.gcode - Copy.zip](https://github.com/bambulab/BambuStudio/files/9642703/Base_plate_1.gcode.-.Copy.zip)

I had to change the .3mf to .zip to get it to upload.

Sebazzz, 2022-10-03:
Another sample project:

[image]

[image]

[halloween.3mf.zip](https://github.com/bambulab/BambuStudio/files/9700439/halloween.zip)

ptsao88, 2022-10-11:
I am also seeing this issue.  See images below.

Bambu Studio seems to be automatically expanding supports, which then do not follow the previously defined support blocker areas, resulting in supports in unwanted places like inside of the model and at the bottom.

[image]
[image]
[image]

ptsao88, 2022-10-11:
You can also see this issue if you paint all faces with blockers, and then use a support enforcer at this specified location.

If the support enforcer covers the whole overhang area - it gets expanded and eventually generates supports elsewhere, illustrating that the software not only expands supports behind the scenes, but also ignores previously defined blockers.

If the support enforcer is shrunk to be smaller than the overhang area you get the desired result...however, the supports now no longer cover the entire overhang, which can lead to drooping perimeters.

[image]
[image]
[image]
[image]

MusangKingBambulab, 2022-10-26:
Thanks for the questions sharing from @BryanHaven and @Sebazzz and the solutions provided by @ptsao88. The reason why the support blocker is not valid is that the support generation mechanism under the default "Grid" mode -  the projected region of support per layer will dilate outward based on the concern of the print path generation following the rasterized grids with different resolution and you can tune this resolution by the "Base pattern spacing" parameter in the BambuStudio: Support->(Advanced) Base pattern spacing (default is 2mm). Unfortunately, the "Snug" mode under normal support is still under development and not open to the question from @Sebazzz. I believe it will come out soon. In addition, "Hybrid (auto)" support will be another good choice for @BryanHaven.

ptsao88, 2022-10-26:
@MusangKingBambulab I have another ticket that is relevant to this here: https://github.com/bambulab/BambuStudio/issues/437

Adjusting "Base pattern spacing" isn't ideal because to prevent dilation, you need to set this to 0mm...which makes support structures 100% infill.  Ideally we can control the dilation independently from support infill percentage.  This is how many other slicers commonly work for support material.

MusangKingBambulab, 2022-10-27:
> @MusangKingBambulab I have another ticket that is relevant to this here: #437
>
> Adjusting "Base pattern spacing" isn't ideal because to prevent dilation, you need to set this to 0mm...which makes [...]

Thanks for your advice, making more customized parameters of support generation open has been added to our plan.

MusangKingBambulab, 2022-11-28:
Dear all @BryanHaven @Sebazzz and @ptsao88 , a new config parameter called "Support xy expansion" for normal support has been involved in the next release. I have tested all the above-mentioned models, and it looks good (check the below figs). Hope you all enjoy 3D printing and provide us with more suggestions to improve the studio.
[image]
For dry box model, please check another ticket #417

ptsao88, 2023-09-07:
These issues have been solved for me by using the new "Snug" support style feature.

DarrenLynch, 2023-09-07:
Yer I saw some comments in the forum on that, unfortunately it does not fix it for me, its better, but the blockers are still not properly respected, and a blocker is a blocker, partial respect is pointless.

ptsao88, 2023-09-08:
Yes blockers still don't work as expected - they're somewhat unintuitive...but the snug feature has solved all the instances where we originally wanted blockers in the first place, at least for us!

factoidforrest, 2024-01-31:
Still happening for me

tangketan, 2024-02-01:
Support blockers only block generating new supports in its area, not block old supports passing through. You guys can use tree supports.

ardugitter, 2024-04-02:
Still happening for me too.