bambulab/BambuStudio #327: [Enhancement] Enable automatic printing order wall/infill when purge to infill is enabled

hutje, 2022-09-25:
When purge to infill is enabled, it would be incredibly usefull to be able to choose to let BambuStudio calculate the best order for each layer to begin with infill or the walls. After cutting the filament, the printer could without purging easily start printing inside the infill to ensure the right color. This way it saves a lot of time and waste material.

SaltWei, 2022-09-26:
Currently, it should print infill first after changing filament.
Could you share 3MF file if you find some order issue?

hutje, 2022-09-29:
@SaltWei I have included the 3mf file... But the purge does not happen in the infill.
Look at for example layer 27 and layer 28.
[Purge not in infill.zip](https://github.com/bambulab/BambuStudio/files/9672367/Purge.not.in.infill.zip)
It would be more logic to start the filament change somewhere inside the last infill and continue to extrude in the infill before starting on the destined walls. Then start the filament change after doing the walls and start infill during the color change. Depending on the purge volumes it could calculate how much infill it needs to print before doing the walls and therefore can switch between starting with infill or starting with walls to be able to use the infill for the next filament change.

Additionally it could give a recommended value of the amount of infill needed to completely get rid of waste filament.

This way it reduces a lot of waste.