bambulab/BambuStudio #329: [Question] X1C Camera

prawnz, 2022-09-26:
Is there any way to link the X1C camera to homeassistant?

bambu123, 2022-09-26:
We have planned a virtual camera feature, to stream the printer video to localhost UDP/RTP/SDP for softwares such as OBS to broadcast the video.
But we are not going to support HomeAssitant directly.

prawnz, 2022-09-27:
If you can stream thru RTSP that should be sufficient.

bambu123, 2022-09-28:
We think RTSP is too complicated and a .sdp file is enough. You can then use VLC or OBS to get the RTSP stream.

prawnz, 2022-09-29:
Won't you be preferring HLS as it will make it easier to be incorporated/change to modern devices?

bambu123, 2022-09-29:
Perhaps you need an FFmpeg server to get all the things. We are just a camera device, not a streaming service.

vilic, 2022-10-14:
Any ETA available?

bambu123, 2022-10-14:
We plan to complete this feature in November.

RobinWhitfield, 2022-10-17:
Documentation for this has started to appear:

https://wiki.bambulab.com/en/software/bambu-studio/virtual-camera

bambu123, 2022-10-18:
Yes, it is not completed.

ImagineerNL, 2022-11-14:
Will you consider making this streaming app work on linux too, so we can have it running on a server (without needing to keep the pc open for streaming purposes)

bambu123, 2022-11-15:
it seems possible, but there is no easy way to config printer URLs across computers.
The streaming tool needs a URL that can only be fetched from Bambu Studio.
Also, it needs to be built from the source code in Linux and we supply only x86_64 binary for the ubuntu like system.

ImagineerNL, 2022-11-15:
I'm quite sure people using linux are able to find the ip address of their printer, if thats what you mean

bambu123, 2022-11-16:
However, the URL is dynamic for security.