bambulab/BambuStudio #331: Extremely slow slicing - further optimisation possible?

aquatacizem, 2022-09-26:
A major issue I've been struggling with in Bambu Studio is that slicing certain files with complex, detailed paths takes many times longer than either PrusaSlicer or SuperSlicer.

Bambu Studio really struggles to generate the gcode, if it can even generate it at all. Some files I work with will not move past 5% even after 24 hours. Bambu Slicer does not freeze or crash and is still actively processing, but with seemingly no progress.

If it is helpful, I have attached a mock-up model I created as stress test. In PrusaSlicer, it took around 14 minutes to slice. In Bambu Studio it took around 40 minutes. 

It would be really great if slicing times were able to be further optimised. Thank you!
[Slicer Stress Test.zip](https://github.com/bambulab/BambuStudio/files/9645346/Slicer.Stress.Test.zip)

SaltWei, 2022-10-09:
I think this is caused by different enabled settings.

SaltWei, 2022-10-09:
Bambu Studio always enable “Detect bridgeing perimeter” and also enhance this setting to calculate the overhang degree for every segment part of perimeter, not only detect bridging perimeter. This is useful and necessary for overhang slowdown feature.

Since Bambu Studio always enable this option in code, user can't disable this option. I disable this option in code and slicing is much faster.

Another option to make gcode generation to be slow is "Reduce infill retraction" which is same and named as “Only retract when crossing perimeters” in prusa. BambuStudio always enable this option, but prusa is not. Enabling this option can reduce a lot of printing time but make slicing slower.

This is my full analysis for this issue.

1 Option “Detect bridgeing perimeter” is enhanced and make slicing slower than prusa, but it's hidden and can't be disabled by user.
2 Option "Reduce infill retraction" is always enabled in BambuStudio but Prusa is not. This can be disabled by user.

So the next work is try to optimize the speed related to these two options, but I have no idea currently because a lot of 2D geometry calculation related to clipper lib need to be investigated.

SaltWei, 2022-10-09:
OK. We find one "solution" for "Conclusion 1". When the model has no overhangs, for example, the model you shared above, user can disable "slow down for overhang", Bambu Studio is unnecessary to calculate overhang degree of every segment of perimeter. This is one point to optimize.

I locally write the code. And disable "slow down for overhang" and "Reduce infill retraction". Re-slice again and time cost is less than 5 minutes, which is much faster than enabling these two options. Prusa Slicer is also same time on my computer.

At least, now we can manually disable these two options to make settings to be same with prusa, and time cost to be close as well.

Will push the patch.

The final optimization is make slicing to be fast even when enable these two options. Not easy to investigate 2D geometry calculation related to clipper lib but we will keep on it.

aquatacizem, 2022-10-09:
Fantastic! Your time and work is much appreciated. Thank you!

SaltWei, 2023-04-18:
https://github.com/bambulab/BambuStudio/releases/tag/v01.06.00.65 has improved the slicing speed a lot for this model even with overhang-slow-sown enabled

closed 2023-04-18 by SaltWei