bambulab/BambuStudio #333: Slicer chosen AMS slot not what comes up in Print dialog box and sent to AMS

warpster, 2022-09-27:
If I load a model, set the slot, slice and hit print, it will have the wrong slot choice selected in the resulting Print dialog box. Now I can change it there and the AMS will use the correct slot but if I do nothing it is set for the wrong slot and will print with the wrong filament. This just started happening two days ago (before upgrade) and only for slot 3 (it will go to slot 4). All the other slot choices remain correct in the Print dialog box. Here is a picture showing the issue...
[image]

YifanWuBambu, 2022-09-28:
Hi @warpster ,
The Studio will auto-search the best matching AMS slot for each filament in the print job.
In this case, the spool in AMS slot 4 is a grey ABS, so it choose this slot.

DeanDavis, 2022-11-18:
Don't do that. Or at least make it a preference I can turn off.
In my slicer I just want to set the filament slots and have the printer use exactly the slots I pick.
Sometimes I don't want to adjust the colors in the AMS or on my slicer project.
I know the colors loaded in the AMS.
My example...
[image]
I've Picked slot one as the base and in this job I have it changing to slot 2 at a certain height.
In the slicer I have slot one as White but in the AMS it is actually a blue filament.
In the slicer I have slot 2 set as red but in the AMS it is actually white filament.
I hit print and it's showing it's going to use Slot 2 (and showing a White color?) and slot 1 (showing a Red) color?

And when it started the physical print it loaded slot 2 for the base. So it was going to start printing in white as the base.
Not what I wanted at all.

All so very confusing.

YifanWuBambu, 2022-11-23:
>> In the slicer I have slot one as White but in the AMS it is actually a ###BLUE### filament.

There seems another problem.
In this picture, the AMS slot 01 is physically loaded with a RED filament. Need to confirm in reality, whether is a BLUE or RED filament? If it is a BLUE one (be wrongly recognized as RED), is it a Bambu Filament or a 3rd party filament?
[image]

Sometimes, you may create a project with more than 4 colors in different plates. In this case, you need to use this AMS slot mapping feature for the 5th, 6th, ... filament.

For bambu filament, if you still think this is confusing (even all filament are correctly recognized), we may add an option to force sequential mapping (slicer1 => ams 1, slicer 2 => ams 2, ...) in preference setting.
For 3rd party filament, there is an easy way to avoid auto-mapping. Just skip setting each filament's property on printer's screen or in device page, and sequential mapping will be used.