bambulab/BambuStudio #336: Time estimates are off since the update

mctazman, 2022-09-28:
Since the last update, the time estimate of the slicer's quite far off. See the attached screenshots.
[image]
[image]
[image]

I'm on version 01.02.00.09 of Studio, Mac OS 12.6 (21G115), latest firmware on the BL

mctazman, 2022-09-28:
Forgot to mention, this issue was not present pre updates.

CrunchBlood, 2022-09-28:
I am finding the same issue.
Started a print last night at 22:05 which was meant to take 11h16 minutes. Expected it to finish at 09:21ish this morning but didn't complete until 11:53am. On the printer screen it said completed and time for print was 11h16. Didn't do screen shots but will do tonight when I run the same print again.

FreeManFreeThinker, 2022-09-28:
I can confirm I am experiencing this as well.

SaltWei, 2022-09-29:
> I'm on version 01.02.00.09 of Studio, Mac OS 12.6 (21G115), latest firmware on the BL

Could you please share this model as well, so that we can reproduce this issue under same condition？

SaltWei, 2022-09-29:
> Forgot to mention, this issue was not present pre updates.

Interesting finding. Thanks for response.

mctazman, 2022-09-29:
> > I'm on version 01.02.00.09 of Studio, Mac OS 12.6 (21G115), latest firmware on the BL
> 
> Could you please share this model as well, so that we can reproduce this issue under same condition？

X1C Combo, in my case.

SaltWei, 2022-09-29:
> > > I'm on version 01.02.00.09 of Studio, Mac OS 12.6 (21G115), latest firmware on the BL
> > 
> > 
> > Could you please share this model as well, so that we can reproduce this issue under same condition？
> 
> X1C Combo, in my case.

Multi-color printing or single?
If is single color, then it should be a bug. If is multi-color, then it's normal because time of changing filament is unstable and can't be estimated correctly when slicing.

mctazman, 2022-09-29:
> Multi-color printing or single? If is single color, then it should be a bug. If is multi-color, then it's normal because time of changing filament is unstable and can't be estimated correctly when [...]

Single colour, multiple prints. Besides, the previous multi-colur prints I did had (more or less) perfect estimates :P You're underselling your previous calculations :D These were spot on

CrunchBlood, 2022-09-29:
Did another last night where it took 2hr30 longer than expected.
[image]
[image]
[image]
[image]

mctazman, 2022-09-29:
@CrunchBlood Which Windows/BStudio/Firmware version? & Which machine?

I'll be posting firmware in a bit, gonna print in a sec :D

mctazman, 2022-09-29:
My firmware versions:
[image]

mctazman, 2022-09-29:
Now printing with BASF petg, though chances are high it’ll not complete due to bad filament

mctazman, 2022-09-29:
Just checked, time on printer's the same as time in slicer & mobile app on iOS

CrunchBlood, 2022-09-29:
> @CrunchBlood Which Windows/BStudio/Firmware version? & Which machine?
> 
> I'll be posting firmware in a bit, gonna print in a sec :D

My Bad,

It was Windows 10, latest firmware V01.02.00.09 , X1 C Combo

Haha print away, I still am just have to adjust my loads accordingly :D

SaltWei, 2022-09-30:
Thank you veru much for these information. Very important for us to investigate.
And this is clear now and it's bug of first layer scan of printer which lower the acceleration but not set back.
Printer firmware will be updated and fix this issue.
Close first layer scanning also can be a workaround.

mctazman, 2022-09-30:
> Thank you veru much for these information. Very important for us to investigate.
> And this is clear now and it's bug of first layer scan of printer which lower the acceleration but not set back.
> Printer firmware will be updated and fix this issue.
> Close first layer scanning also can be a workaround.

Cheers.

SaltWei, 2022-10-11:
Done in printer firmware. Will be included in next release.