bambulab/BambuStudio #337: X1C Camera exposure/contrast controls etc

WanaGo, 2022-09-28:
Is there any way to add in controls for the camera, for exposure/contrast/brightness type things?
Its incredibly dark even with the light on, and its really hard to see.
It actually feels like it got worse after the last update, but I may be mistaken.

If there was a way to change the camera output so you can brighten it up a bit etc, that would be really helpful.

sodachen22, 2022-09-30:
1. There is no way to set exposure/contrast/brightness for chamber camera. The policy is AE is used when there is no timelapse, and the exposure will be fixed when timelapse started.
2. We found a bug when exposure is fixed or released, which may lead the picture is too dark or too bright. The bug will be fixed in next firmware of printer release
3. Please do not turn off charmber light when timelapse is enabled.

WanaGo, 2022-10-18:
To add to this, if I turn on or off the office lights once a print has started, the timelapse is hugely affected, so much so that it can be way over exposed and too bright, and the camera does not compensate for the change in lighting. This is just by turning on or off the room lights.
At the same time, I then start to get false spaghetti warnings etc, as obviously the camera exposure is so wrong when the lighting changes, that it thinks it sees spaghetti, when it is in fact printing totally perfectly still.
Long prints mean the lighting needs to be able to change, as we do not leave the lights on or off the entire time, as we have to use the office for working, and we turn it off when we leave for lunch or for the end of the day, and if the print is still happening then the camera should be able to handle this situation - surely.

CCS86, 2023-01-12:
This is definitely an issue.

Bambu Studio's video display is significantly darker than the video in Bambu Handy.

I would like to have both an actual gain control on the camera, plus brightness & contrast for the video display in BS.

sodachen22, 2023-01-13:
@CCS86 Could you please check quality of the video recording file in SD card? The timelapse recording, video recording, liveview to Studio/Handy are frome the same input and same output of codec. They should be the same. The difference is APP and Stduio have difference decoder and display.

CCS86, 2023-01-13:
> @CCS86 Could you please check quality of the video recording file in SD card? The timelapse recording, video recording, liveview to Studio/Handy are frome the same input and same output of codec. They should be the same. The difference is APP and Stduio have difference decoder and display.

Great question.

I can verify that a timelapse video does have pretty good exposure. Significantly brighter than the live feed in BS. So, the issue is probably in that player codec.

sodachen22, 2023-01-14:
@CCS86 Studio does not include codec, it uses system media player interface. Maybe the setting of the OS needs to be checked. If you share the OS information here, maybe others who have the same OS can check where the difference is.

CCS86, 2023-01-14:
> @CCS86 Studio does not include codec, it uses system media player interface. Maybe the setting of the OS needs to be checked. If you share the OS information here, maybe others who have the same OS can check where the difference is.

That's odd, since when I play back the recorded time lapse on that PC, the brightness is fine.

Here are the OS build details:

[image]

Jaerin, 2023-04-26:
A setting to allow for the adjustment of the exposure or a setting to have it adjust the AE even when timelapse is enabled should be added.  Having a fixed exposure at the start of the print when it may take multiple days and have changes in chamber light between night and day.  I don't think anyone thinks the dark underexposed timelapses are useful.  Also wouldn't this affect the print failure detection if the camera is too dark?

LightningXCE, 2023-05-18:
Echoing everything else here; If anything, can we have a setting to be able to adjust the locked-in exposure setting? I use extra lights in my setup and this would allow adjusting it properly. Even just having the setting to manually set that "locked" setting once it's started would be a game changer for visibility.

megamaniac, 2024-01-26:
Adding my own voice to this. Please allow manual exposure control. It would be a massive help for checking on prints (especially dsrk/black prints)

megadude9704, 2024-02-18:
PLS

CCS86, 2024-04-06:
The camera AE seems dead set on underexposing!

Even after adding some additional LED lighting, and the build area looking very bright to the eye:

[image]

Here it is in Bambu Studio:

[image]