bambulab/BambuStudio #338: ignoring build plate only support

PowerChaos, 2022-09-29:
hi all,
this is just a quick bug report , i was slicing my own file and i used auto support
when i checked the file i saw that there was support going to be printed inside a closed print
this means it just ignore the setting, 

The file can be provided , it can easy be reproduced as it keeps doing that ( for my file atleast ) 
also they start at layer 10 or so , so thats real high from the "build plate only " place 
i saw some other glitches to , like manual support , it is not painted but tree support still love to place there a tiny bit of support
it also seems that it is not always the case , now that i am generating it a couple of times to get the screenshots

in this case it is a critical bug, as it is a closed object that transfer liquid, and it wont really move smooth if there is a support wall inside that blocks the flow

[image]
[image]
[image]

this last screenshot is to give the overview of where it happends
but i can not reproduce that specific screenshot anymore , maybe it depends where i place it on the build plate , but it still shows that it is not correct
[image]

if you need the file , please let me know and i attach it to github

Greetings from PowerChaos

candre23, 2022-10-06:
I'm noticing the same problem.  It seems to only affect normal supports.  Even if you manually paint support blocker on these bridges, it will still support them.

[image]

tangketan, 2022-10-08:
@PowerChaos this is a bug with hybrid(auto) support. We have fixed it, please wait for next release.

tangketan, 2022-10-08:
@candre23 Sorry but are you talking about another bug? Because the picture you posted shows normal support type.

PowerChaos, 2022-10-12:
@tangketan 
got also a new addition to this bug
but this time the other way around
ignoring forced supports
as you can see in the image , it shows the supports on one side ( both sides ) but ignore the other sides ( both sides )

most likely related to the same bug , but can not confirm it yet as the new slicer is not released yet when i made this report

Greetings from PowerChaos
[image]
[image]