bambulab/BambuStudio #342: [BUG]: Filament slots between prints somehow "lost"

attilah, 2022-09-29:
Here is what happened:

AMS Slot 1: PLA
AMS Slot 4: Bambulab Support W

Printed a model (20 hours) with supports and separation layers, Support filament assignment was 1 and 2 respectively

Added a new plate to the same project after print, loaded a new STL sliced it and sent to print. In the send dialog filament slot assignments were correct, 1 and 4 as previously.

At the start of the print I got an error message that AMS Slot 2 has stuck filament, etc...obviously it tried to use the wrong slot for some reason...

Canceled the print, rechecked everything, sent it again, same error...

Power cycled the machine, sent it again, works as expected. Notice NO changes were done anywhere.

X1C, latest firmware, latest slicer.

HobinBambu, 2022-09-30:
@attilah 
Hi attilah,
could you please create a ticket and share your printer log and studio log,here are some reference links.
Thanks!
https://wiki.bambulab.com/en/x1/troubleshooting/upload-log
https://wiki.bambulab.com/en/x1/troubleshooting/bambu-studio-log
https://wiki.bambulab.com/en/x1/troubleshooting/export-failed-print-log-to-sd-card

attilah, 2022-09-30:
Thanks @HobinBambu opened the ticket.