bambulab/BambuStudio #343: Seam placement

kyle1046, 2022-09-30:
Suggestion for additional seam placement option.
"Sharpest Corner" this way the seams are practically invisible instead or seeing a line of seams or seeing them at random throughout the print

YifanWuBambu, 2022-09-30:
Hi @kyle1046 ,
Could you provide one or several sample models for this?

kyle1046, 2022-09-30:
> Hi @kyle1046 , Could you provide one or several sample models for this?

If you throw a few models into cura and select sharpest corner you'll see that all the seams are placed in the sharpest corner that is can find for the layer, so when it's printed the majority or the seams are hidden and gives the appearance that the model wasn't printed

YifanWuBambu, 2022-10-04:
In Bambu Studio, although the default seam position names "Aligned" or "Nearest", it also considers the sharp corner (both concave and convex), and tends to place the seam there.
This WiKi article introduces the specific logic.
https://wiki.bambulab.com/e/en/software/bambu-studio/Seam