bambulab/BambuStudio #344: no video feed in bambu studio - "Missing BambuSource component registered for media playing! please reinstall bambu studio or seek after sales support"

ryanmcgoldrick47, 2022-09-30:
Tried uninstalling and reinstalling, tried older and latest version of software.

Trying to view the video feed of printer that is at home from an install of Bambustudio on my work laptop at work.

bambu handy on iOS works with video feed, and bambu studio at home works with video feel but not on my work laptop.

Windows 10
Dell precision 7560 laptop

ryanmcgoldrick47, 2022-09-30:
[image]

bambu123, 2022-09-30:
Maybe BambuSource.dll is blocked by overly sensitive antivirus software.

Here is the FAQ about the details.
https://wiki.bambulab.com/en/software/bambu-studio/faq/live-view

ryanmcgoldrick47, 2022-09-30:
@bambu123 
I went to the IT admin, we turned off all anti virus software for 4 hours unintalled app, reinstalled app. Same error message appears, I checked the Appdata Plugins folder for Bambusource.dll and the file is indeed there, I also checked the security permissions and it has full permissions for everything. so can confirm it is not antivirus software as also checked event logs for antivirus and nothing came up before we disabled it.

ryanmcgoldrick47, 2022-09-30:
[image]

[image]

bambu123, 2022-09-30:
Please check the registry. Open Power shell or cmd.exe, and execute the following scripts:

REG QUERY HKEY_CLASSES_ROOT\bambu
REG QUERY "HKEY_CLASSES_ROOT\CLSID\{233E64FB-2041-4A6C-AFAB-FF9BCF83E7AA}\InProcServer32"

bambu123, 2022-09-30:
Make sure not to install BambuStudio with the Administrator account, and use it with another Windows user account.

ryanmcgoldrick47, 2022-09-30:
> xecute the following scripts:
> 
> REG QUERY HKEY_CLASSES_ROOT\bambu

[image]

ryanmcgoldrick47, 2022-09-30:
> Make sure not to install BambuStudio with the Administrator account, and use it with another Windows user account.

everything you install on work computers require administrator to install, we cannot install without them logging in for elevated permissions

ryanmcgoldrick47, 2022-10-02:
@bambu123 
[image]

ryanmcgoldrick47, 2022-10-02:
[image]
@bambu123 haven't tried running powershell or cmd as admin yet as need to be back in office infront of IT so they can login as admin to try
but doing it myself thats what I get.

bambu123, 2022-10-02:
The error code says to have none permission to register the dll. it is because something block it. maybe anti-virsus or some secure strategy.

ChrisG-hub, 2022-10-02:
Just run the installer with administrator rights.
[image]

ryanmcgoldrick47, 2022-10-02:
@ChrisG-hub thanks for the suggestion but we already tried doing this and same error happens

ChrisG-hub, 2022-10-02:
Well, i would use an old fashion elevated command prompt to try to register the dll.
%systemroot%\SysWoW64\regsvr32 %APPDATA%\BambuStudio\plugins\BambuSource.dll 
or 
use the 32-bit version of Regsvr32 to register a 32-bit DLL on a 64-bit version of Windows If you receive an error when registering a 32-bit DLL on a 64-bit version of Windows, follow these steps:

Open an elevated command prompt.
Copy or move the **BambuSource.dll** to the %systemroot%\SysWoW64 folder.
%systemroot%\SysWoW64\\regsvr32 **BambuSource.dll**
Info:
The 64-bit version is %systemroot%\System32\regsvr32.exe.
The 32-bit version is %systemroot%\SysWoW64\regsvr32.exe

Pmant, 2022-10-03:
Same issue here but with the portable (zip) version. On first start it installs the networking plugin which is working fine but the video plugin does not work.

ryanmcgoldrick47, 2022-10-03:
@ChrisG-hub Tried all of the steps you suggested above but nothing changes, still same error in bambu studio, and I get this when entering that top code you wrote, even after moving the .dll manually into both the 64 and 32 bit version. 
[image]

bambu123, 2022-10-08:
> [image] @bambu123 haven't tried running powershell or cmd as admin yet as need to be back in office infront of IT so they can login as admin to try but doing it myself thats what I get.

The power shell can't expand %APPDATA%.

bambu123, 2022-10-08:
> @ChrisG-hub Tried all of the steps you suggested above but nothing changes, still same error in bambu studio, and I get this when entering that top code you wrote, even after moving the .dll manually into both the 64 and 32 bit version. [image]

No, do not register BambuSource.dll with the 32 bits regsvr32. It is 64 bits binary.

bambu123, 2022-10-08:
Secure strategies must have blocked the register.

bambu123, 2022-10-09:
> Same issue here but with the portable (zip) version. On first start it installs the networking plugin which is working fine but the video plugin does not work.

The BambuStudio installer will register BambuSource for video playing. If you use the zip package of BambuSource, you should register BambuSource using regsvr32 in cmd.exe. Also, you need to install the fonts and the edge runtime.

bambu123, 2022-10-31:
A new version of Bambu Studio will retry register BambuSource. You can try it to see what happens.

bambu123, 2022-11-17:
@ryanmcgoldrick47 please help try the new version of Bambu Studio.

bambu123, 2023-02-03:
@ryanmcgoldrick47 Have the problem been solved?

SaltWei, 2023-03-02:
v1.5.0 public beta has been released. Please let us known if still exist.
Thanks!

closed 2023-03-02 by SaltWei