bambulab/BambuStudio #347: Inconsistency, gaps on top layer

CptHungarian, 2022-09-30:
If there are small details on the top layer, typically numbers or letters Bambu Studio can't seem to handle them properly.
Instead of filling out the top layer consistently, it creates internal walls and gap fills around and under the letters and numbers.
This is a good example (screw gauge by Bambu Lab):
[image]
[image]
The same STL sliced with PrusaSlicer 2.5.0:
[image]
[image]

SaltWei, 2022-10-09:
This is same behaviour with superslicer which caused by "only one wall on top surface". Disable this option can be a workaround.

Should be improved in the future.

CptHungarian, 2022-10-09:
> This is same behaviour with superslicer which caused by "only one wall on top surface". Disable this option can be a workaround.
> 
> Should be improved in the future.

Yep, that fixes it. Thank you!

SaltWei, 2022-10-09:
You are welcome!

Disable this option is just workaround. Should be root fixed to be normal even when enable this option.
We will keep on it further.

AbeFM, 2022-10-30:
Came to report the same issue, and the same work around. This used to slice fine on PrusaSlicer - but that was without the one wall option even available.

djngnear, 2022-12-21:
I thought maybe your issue was similar to mine.

https://github.com/bambulab/BambuStudio/issues/708