bambulab/BambuStudio #348: Flow Calibration

CptHungarian, 2022-09-30:
I tried to get answer to these questions several times but couldn't get any answer.

What does the flow calibration do exactly?
Does it actually adjust the flow ratio or is it just a linear/pressure advance calibration?
Where is it stored and more importantly how long is it being stored?

Here is an example of my recent prints where I printed the top part without any calibration.
Then I decided to do a "flow calibration" and the bottom part suffered from washed out details, rounded edges, etc.
Exactly what bad linear/pressure advance calibration would cause.
[image]
[image]

The next day I reprinted the same part without running the calibration and all these problems disappeared... (apart from the terrible ringing, but that's a different topic)
[image]
[image]

My theory is that the printer only stores the linear/pressure advance calibration values temporarily and these values are only valid for the current print.

steezyprintz, 2022-10-02:
I would also like to know more about the flow calibration is stored etc. I am assuming its stored in its memory until another one over writes it at the moment as it seems to be happening to mine. 

What I have learnt recently with my X1C is that the colour of the filament does make a noticeable difference to how well the flow cal works and then how well prints come out. For example dark colours like black and grey they too much like the build plate and I was getting errors/warning about exposure from the lidar. Adding more light didn't help much, So I changed to yellow in the same material / brand and the z seam and the corners sooo much better for it. I ran the flow cal and printed a small box in yellow, I then switched to printing what I wanted in black same material and brand (esun PLA+), print came out very nice after that in black. I am pretty sure I read about this on the facebook page several weeks ago from someone experiencing the same.

I would really like the flow calibration to be a displayed value that can then be added to custom/3rd party material settings to avoid needing to do that process too many times. Being able to set it up would be ideal as bambu labs range of materials is rather limited currently.

I would really like Bambu labs to sell the RFID tags that we can stick onto 3rd party materials spools and use it like their own, storing the flow cal on to the RFID tag, how much used, the colour and material type etc.

psiberfunk, 2022-10-04:
I would suggested @wuyifanfancy , that perhaps when you get back from the holiday, that this ought to be tagged as a bug ?  It's either a bug that you say you do flow calibration and you don't (e.g. rename what you call it)... or it's a bug in the sense that flow calibration SHOULD be happening, but does not.

steezyprintz, 2022-10-05:
yeah so the translation of flow calibration or the meaning of it is like you say @psiberfunk I don't think anyone here was misunderstanding it to mean flow ratio.

I would love for the flow ratio to be measured and adjusted for automatically as well as other filament dependent settings like the shrinkage, if bambu labs can do this they will make me very happy but also move home 3D printing actuate parts for n00bs so much further forward to being the magic plug and play!

psiberfunk, 2022-10-05:
I personally actually did misunderstand it to mean flow ratio from the marketing videos and such until I ran my tests on my new printer !

Liamnerf, 2022-10-08:
Hopefully Bambu lab, in future, can modify their code in order for the LiDAR to calibrate flow as well linear advance. Perhaps when they develop a standalone flow calibration process. Presumably any flow ratio settings set by the user in Bambu studio would need to be disabled.

CptHungarian, 2022-10-15:
@YifanWuBambu So can someone from Bambu answer these questions please? To be honest, this information should be publicly available for all users either in the wiki or user manual, as this is one of the main selling points of the printer.

psiberfunk, 2022-10-24:
This information is now getting misrepresented by Bambu's own wiki team ! Here: https://wiki.bambulab.com/en/x1/manual/manual-flow-rate-tuning it suggests that "The X1 is able to automatically calibrate the flow rate based on the filament used"  but that you can do better by "tuning it manually".  The reason is because.. shocker.. it's **not tuning the flow ratio at all** as the wiki page explains how to do manually.

It's not a good thing when Bambu's own people are conflating the difference between flow ratio and linear advance.  Could Bambu please address this confusion before it gets worse @YifanWuBambu  ?

RobinWhitfield, 2022-10-25:
> The reason is because.. shocker.. it's **not tuning the flow ratio at all** as the wiki page explains how to do manually.

Are you 100% sure this is the case?

The flow rate provided in the slicer could very easily be a multiplier on top of automatic calibration.

If there is no flow rate calibration happening, then surely the pattern printed to the left of the PA lines would be totally superfluous.

Merlwynd, 2022-10-25:
I would also like to get clarity or confirmation on this. Important simply to know limitations so one can calibrate properly using common tools (i.e. hollow calibration cube). Doesn’t look like anyone at Bambulab yet assigned to provide answer?

psiberfunk, 2022-11-03:
@YifanWuBambu  :  What should we do with this one ?  It seems like it may be improperly categorized... does this need better documentation, a software fix, or what?

steezyprintz, 2022-11-03:
My vote is for both, documentation improvement to explain it cheaper and software improve for adding in the flow calibration and over rides for different lines types please. 

Currently for me I am having to adjust the flow of materials so I get a good top surface but that then means my walls are to thin and also my bottom layers aren't nice too.

YifanWuBambu, 2022-11-04:
@CptHungarian @psiberfunk @steezyprintz @RobinWhitfield 
Bambu Forum is recently online. Firmware related issue  can be response much better there.
https://forum.bambulab.com/c/bambu-lab-x1-series/16

-------------
For Bambu Studio issue or feature, we still discuss on the github.

psiberfunk, 2022-11-04:
Do we consider this to be firmware or slicer related @YifanWuBambu  ?  It's sort of in the grey "in between zone" ?

YifanWuBambu, 2022-11-08:
@psiberfunk 
This is actually a firmware related feature, not in the grey zone in between.

Currently, the flow calibration is a "dynamic" adjustion of flow, which works like "pressure advance".

psiberfunk, 2022-11-08:
Ok , so if it’s not slicer related , where should we open a ticket for this to be properly documented ? The community forums you suggested are not a ticket-tracking mechanism ..?

RobinWhitfield, 2022-11-08:
@YifanWuBambu, can you confirm whether or not some kind of documentation of this kind of firmware feature is being worked on?
It doesn't need to be anything fancy.

Right now the true potential of this machine is being limited by the lack of documentation for many of its features.

Documentation on how to use the M900 command, or the camera commands, etc. are going to help a competitor with their product, but it would sure help us X1 **owners**.

H4kunaM4tata, 2022-11-08:
> Ok , so if it’s not slicer related , where should we open a ticket for this to be properly documented ? The community forums you suggested are not a ticket-tracking mechanism ..?

+1 to this

I opened a firmware issue here with the printer overwriting the vitrification temp with warning, it was suggested to open a ticket on that forum.

Yo, the forum has almost no activity imo, looks new asf, GitHub is still the best place for general issues.

We aren't getting enough support for issues like this, lack of Linux client is perfect example of total lack of BambuLab support here, what to say a brand new forum?!

Not gonna happen.

CptHungarian, 2022-11-10:
> @CptHungarian @psiberfunk @steezyprintz @RobinWhitfield Bambu Forum is recently online. Firmware related issue can be response much better there. https://forum.bambulab.com/c/bambu-lab-x1-series/16
> 
> For Bambu Studio issue or feature, we still discuss on the github.

Okay, I posted the same questions there.
I'm not surprised that no one answered them yet.

https://forum.bambulab.com/t/flow-calibration/892

psiberfunk, 2022-11-10:
Yea it’s really an issue when there’s not actually an ISSUE tracker for the firmware .  The community forum approach doesn’t substitute for work tickets …

psiberfunk, 2022-11-13:
Still nobody from Bambu has replied to the post in the other forum... ?

H4kunaM4tata, 2022-11-13:
> Still nobody from Bambu has replied to the post in the other forum... ?

That forum is useless, new, has barely any post and if they are unable to provide update here where it's so easy to track issues/update regarding a feature that is not what they sold, that forum certainly won't solve anything.

SaltWei, 2022-11-17:
We have reported this issue to printer firmware team and this is reply from firmware developer.

Firmware developer:
In our road map , "flow calibration"have " has two parts , dynamic flow compensation calibration and absolute flow compensation calibration.
Dynamic flow compensation calibration : similar to "Linear advance" in Marlin , but added parameters to compensate the non-linearity and hysteresis of melted polymer . Dynamic flow compensation is used to compensate for the lag in the flow respond of the extruder , to make the flow synchronized with the toolhead movement. If you want to make the compensation being linear as the same as Marlin , you can modify the start-gcode , before "M900 Kx.x" cmd , insert a line of "M900 L1000 M10" , after this line is execute , the compensator will work in linear mode when executing the following "M900 Kx.x" .
Absolute flow compensation calibration : this one is used to calibrate the absolute flow ratio , "how much angle should the extruder motor rotate for 1 mm^3 of filament" . This parameter is much harder to calibrate than the dynamic one . For example , 1% of calibration accuracy needs about 2um absolute accuracy ----- although this can be achieved by filtering a lot of data , but in real application , under arbitrary filament , arbitrary temperature , arbitrary bed stains , the calibration needs to be working normally ----- even when it is abnormal it has to ensure that it won't make things worse than without calibration . This needs a lot of testing and debugging , so this function will be provided in the future .

SaltWei, 2022-11-17:
By the way, we also report the situation about no usefully printer issue tracker to printer firmware team.
This is not studio work, but this is Bambu Lab's work.
Hope things can be improved, so that people can report printer issue and suggestions on other forum.

At last, thank everyone to give suggestions to make us keep improvement! This is very very important and really appreciate it.

psiberfunk, 2022-11-17:
A suggestion for the firmware team : offer the ability to calibrate absolute flow by printing a multi-layer (~4-5 layers) test pattern and then inspecting the TOP layer .  This would make the problem obvious to the laser (or maybe even RGB camera for certain colors?) likely.

   This is slower, but the method that the human eye uses , for example, to look for gaps and other inconsistencies .   Trying to do it only on one layer seems like trouble as the firmware guy mentions .  I would suggest to them that since it is rare , it could be done once per filament spool or brand/type instead of every print .

you could make this workflow both relatively painless AND accurate/functional .

psiberfunk, 2023-01-05:
Note @SaltWei and friends:  Your friends over in SoftFever's fork are more or less singlehandedly beating you guys to the punch here.. maybe take some inspiration from downstream and pull in some of those changes?

SaltWei, 2023-01-05:
> Your friends over in SoftFever's fork are more or less singlehandedly beating you guys to the punch here

Thank you for you remind. We had notice this. We will refer to his contributions. Appreciate!

By the way, we are trying to implement a new solution combined with firmware team to make calibration process and data be organized more easily. Please be patient.

Alfiegerner, 2023-04-17:
Hi @SaltWei , do you have any news on the absolute flow rate feature?

psiberfunk, 2023-05-15:
This one is quite old but still very relevant .  I keep getting confusion and questions to this day on it .  Are there plans to fix this ?