bambulab/BambuStudio #350: XY shrink compensation

tapir, 2022-09-30:
Cura, IdeaMaker, SuperSlicer and Simplify3D (via global scale settings) all have this implemented.

Basically the percentage given would scale the model in the X and Y dimensions (preferably independently) giving user control over fine tuning shrinking while cooling down.
Should be under filament settings as different materials behave differently.

I just did a test with this model https://vector3d.co.uk/product/vector-3d-printer-calibration-and-test-suite and I can confirm that ABS prints come up at least 0.65% smaller than intended dimensions.

One can always scale the model manually but having this as a filament setting is the ideal solution

CptHungarian, 2022-09-30:
I think this has been requested already. See https://github.com/bambulab/BambuStudio/issues/314

tapir, 2022-09-30:
Those properties also exist in other slicers along with shrink compensation. They are different.

ptsao88, 2022-10-15:
I would also love to see this available as a filament specific parameter.

I also calibrated my machine yesterday and found that ABS parts come out slightly undersized.

It's a percentage based shrink rate, so a nominal compensation values don't work.  Smaller parts deviate less than larger parts due to the way the polymers shrink after cooling.

I'm currently having to manually non-uniformly scale by 100.5% in both the X and Y axes.  I've been having to do this for years on Ideamaker, so it's not that big of a deal, but it would be nice if Bambu Studio had this integrated properly in the filament settings.

cryd-s, 2022-12-29:
i hope they will add shrink compensation for abs

rwoldberg, 2023-01-13:
I would like this feature added to the filament settings as well.  It can be accomplished by scaling your model every time you import it, but it would be so much simpler if it could just be part of the settings.

brandonraineri, 2023-01-20:
We need this!  It must be added to the filament settings as it will change per filament and temperature.  It needs to be a scale factor, NOT a static XY dimension (X-Y hole and contour compensation in BS) as shrinkage is a % of change.  It is way to tedious having to manually scale each object before slicing and is way to easy to forget to change one of the values.  Please, please, please add this to your next release!

brandonraineri, 2023-01-20:
> 

> 
I actually ended up with the same scale factor (100.5) for XY when using ASA.

We need this! It must be added to the filament settings as it will change per filament and temperature. It needs to be a scale factor, NOT a static XY dimension (X-Y hole and contour compensation in BS) as shrinkage is a % of change. It is way to tedious having to manually scale each object before slicing and is way to easy to forget to change one of the values. Please, please, please add this to your next release!

tapir, 2023-02-24:
Are there any plans to implement this? It's really cumbersome to print fitting ABS parts (bearings, pins etc) with BambuStudio due to this reason. I have to constantly remember to scale the parts or even worse I have to integrate it to my designs which wouldn't work with any other slicer that has this feature implemented.

KiefDelicious, 2023-09-28:
This is the main reason I use Orca slicer (for ABS and ASA). They have it just implemented in the filament profile. The fun thing is, this setting “shrinkage” is even synced via Bambu profiles between computers. I rather use Bambu studio but dimensional accuracy is very important for me.

brandonraineri, 2023-09-28:
> This is the main reason I use Orca slicer (for ABS and ASA). They have it just implemented in the filament profile. The fun thing is, this setting “shrinkage” is even synced via Bambu profiles between computers. I rather use Bambu studio but dimensional accuracy is very important for me.

The wording was a little bit confusing how it worked.  At first I thought all it did was adjust the perimeter width and figured it wouldn't work for large objects.  After playing around with some values, I did confirm this changes the overall scale of the object.  I was also curious if it would get pulled into BS and you confirmed it does.

Not sure why Bambu won't implement this into BS.  It is essential to create dimensionally accurate parts and this seems like such a simple solution.

[image]

tapir, 2023-09-28:
> lemented in the filament profile. The fun thing is, this setting “shrinkage” is even synced via Bambu profiles between computers. I rather use Bambu studio but di

Indeed, I was the one who suggested it and he implemented in 10 days whereas bambu doesn't even reply: https://github.com/SoftFever/OrcaSlicer/issues/360

stony2k, 2024-02-03:
I agree. My X1 prints PLA with exact 0.2 mm undersize. When scaling the part i. X and Y to 100.12 % the real part ist about 99,99% to the real measures, which it good enough. Some shrinkage option is much appreciated instead of scaling every part every time.

MartinChan3, 2024-10-30:
https://github.com/bambulab/BambuStudio/releases/tag/v01.10.00.81 has been released!And now you can shrink as you want.

emberprototypes, 2024-10-30:
Yes, finally! 🎉

MartinChan3, 2024-11-04:
@QingZhangBambu  Now we can close this issue