bambulab/BambuStudio #352: Feature request Live Flow Control

CNCXPROUK, 2022-10-01:
Ability to change the flow rate on the fly.

YifanWuBambu, 2022-10-04:
Hi @CNCXPROUK ,
Could you help explain in what case you will use this feature?

CNCXPROUK, 2022-10-04:
Well traditionally on marlin etc printers you can tweak the flow control if you think it’s too much or too less. I’ve recently been printing with an abrasive filament and it was a real pain to get it to stick on the first layer. Its a pain to keep reprinting something to tune the flow for the first layer whilst everything else is adjustable. Even though I had auto calibrated the flow; Whilst I could change the hot end temp - when the filament isn’t sticking to the bed and just bunching around the nozzle, the ability to see if flow was the issue live would have been useful. 

You are always going to get some scenarios where filament doesn’t behave like it says on the website and the help tutorials don’t apply to that filament.