bambulab/BambuStudio #353: Feature Request Ability to adjust Temperature for individual filaments while print is going

sozo13, 2022-10-01:
I Started printing multicolor Prints on the X1C and AMS and while I am useing all the same type "PLA" I am useing several different brands at once ..and some require different temps , it would make it much easier to tune these in if once the print is started I could easily adjust the temp for a single AMS slot if i see its too hot (stringing) ect  .. I found no lasting way to do this , Yes I could change the temperature WHILE it was useing say slot 1 filiament  ..but once it does a color change and comes back to that color in slot 1 ..its back to useing the temperatures set when the file was sliced.
This would be a Great feature to be able to modify the temp on a single filiment and it last for the rest of the print ..if this feature is here and im unaware sorry , ive just had no luck finding it

YifanWuBambu, 2022-10-04:
@sozo13 
If you want toi apply different temperature for different brand filaments, a suggested method is that you create a filament preset for each filament. Then select your filament preset accordingly.
[image]

sozo13, 2022-10-04:
Yes thanks I figured that out and ot works great , but sometimes I find out
as the print is in process that my preset oa a bit to high and I want to
adjust it down a few degrees mid print based on what  I'm seeing is that
currently possible while the item is printing ?

Thanks  love the x1 carbon

YifanWuBambu, 2022-10-09:
Hi @sozo13 ,
Currently, it is hard to implement this. We did not bind each M104 (G-code to set nozzle temp) with an AMS slot id.
The firmware just simply execute the existing G-code commands one by one.