bambulab/BambuStudio #357: [Feature Request] Option to manually stack objects on build plate

Frontier314, 2022-10-02:
Ability to turn on/off the automatic object placement on build plate.
This would allow us to manually stack objects together for multi-material prints.

ChrisG-hub, 2022-10-02:
Snap and lock at layer height.

coat64, 2022-10-03:
Right click on object > add part > load... The loaded part can be stacked on the original. Set X and Y to 0 to align the centers (or don't) and the Z to your value.

YifanWuBambu, 2022-10-04:
@Frontier314 
One method is as what @coat64  said.
Another method is selecting all objects that you want to stack and right-click and choose "Assemble". All object will be assemble to a big object and each object is considered as a part of that object.

Then, you may move/rotate each part individually. There are two ways to select a part.
1. Alt + Left click
2. Select a part in the object list on the sidebar
[image]

Frontier314, 2022-10-04:
@YifanWuBambu, @coat64 Thank you very much for the suggestions.
What I did until now was stack the objects on PS and then export the STL to Bambu Studio, which it was somewhat inconvinient.

closed 2022-10-04 by Frontier314

YifanWuBambu, 2022-10-09:
@Frontier314 
You are welcome.

If you stack STLs on PS also in this way, you can save it as a 3mf in PS, then import the 3mf to Bambu Studio. The stacking relationship will be kept.
[image]