bambulab/BambuStudio #359: [Feature Request] Make ensure_vertical_shell_thickness and more quality parameter user configurable

mrcg, 2022-10-03:
Please add ensure_vertical_shell_thickness parameter to expert settings.
This is very essential parameter to eleminate unneeded solid infill which can dramatically reduce overall print time and also improve print quality.
Beside this i would like to have much more PrusaPrinter parameter available in expert settings. Especially all quality related parameters.

Quality related parameters available in PrusaPrinter to be also made available in Bambu Studio (user configurable):

ensure_vertical_shell_thickness
extra_perimeters
avoid_crossing_perimeters
avoid_crossing_perimeters_max_detour
detect_thin_walls
thick_bridges
overhangs

Attached screenshots are demonstrating the impact by enabling/disabling the ensure_vertial_shell_thickness parameter:
It's really very essential to have this parameters user configurable.

[image]

[image]

tokamac, 2022-10-04:
"Avoid Crossing Perimeters" is a duplicate of #106
"Detect thin walls" is already in BambuStudio.
"Overhangs" settings are already in BambuStudio, unless you meant a different thing?
I don't know for the others, I think you're right and most of them are not implemented.

mrcg, 2022-10-04:
I have listed all quality related parameters which are bundled in PrusaSlicer in in listed parameter package.
The idea was to reduce effort on implementing this important parameters and avoid asking for each individual parameter seperately. The most important and missing parameter is "ensure_vertical_shell_thickness" but also other parameters should be made available to be user configurable. 

[image]

tokamac, 2022-10-10:
I concur with **Ensure vertical shell thickness**. I have a problem on my X1-Carbon currently because it is ON by default in BambuStudio and it cannot be disabled. I have already enough perimeters (three) on my model, thanks. The toolhead makes unnecessaryaditional weird movements to add this extra 4th inner perimeter, resulting in adding 4 hours to the print time!

SaltWei, 2022-11-29:
Ensure vertical shell thickness has been added. Please wait for next release

mrcg, 2022-11-29:
> [image]

Thanks for reply. If most important parameter "ensure vertical shell thickness" will come in next release i am happy.
The other two left our are not very important.
Looking forward to the next update which then will allow to create pretty clean perimeters for optimized fast printing.

mrcg, 2023-01-17:
FRQ closed now because feature meanwhile implemented.
Thanks to Bambulab team for implementation.

closed 2023-01-17 by mrcg