bambulab/BambuStudio #360: [Feature Request] Edit machine limits in Bambu Slicer

Argolein, 2022-10-03:
Right now it is not possible to edit the machine limits in Bambu Slicer. The options are greyed out. 
If you want to do so you have to edit it via .json file at the moment.

YifanWuBambu, 2022-10-04:
Hi @Argolein ,
Why do you want to edit the machine limits? We gray them out because the default values are consistant with the physical design of Bambu printers.

Argolein, 2022-10-04:
@YifanWuBambu 

Hello,
to answer this it would normally mean open one or two another issues on GitHub because there are two problems I'm going do describe.
**Short answer:** to improved print quality. Lower the effect of ringing and have sharp corners (pressure advance).

**Long answer:**
I've also opened a support ticket in the app but it's more comfortable to post it here. 
Trouble ID: US220930300002

**The problem:**
At default - acceleration 10000 mm/s -printed objects show severe ringing. Support checked my logs and said that vibration compensation is working but is not showing the effect it should. My printer is sitting on a heavy brick stone and I recalibrated it several times.
I let also other user on the discord confirm this as they showed me their prints which also showed ringing:
[image]

It was suggested to me from the support to lower acceleration. I did that and the ringing became better. 
Result of the print by lowering acceleration from 10000 mm/s to 7500 mm/s:
[image]

But by lowering acceleration the edges began to showed bulging which indicates pressure advance is set too low.
I'm always using Flow calibration before every print and I let also others on discord confirm the issue:
[image]

My theory was that the Flow calibration which does also calibrate pressure advance is not using the acceleration I'm using to print. So if you set your printing acceleration to 7500 mm/s in BambuSlicer the printer would not use these values for the calibration. I've checked the generated gcode file and the M204 gcode with the used printing acceleration (you set in Bambu Slicer) is used **after** the calibration. I've only found a M204 before the calibration which is using the machine limits. 

To check my theory I've lowered the acceleration (and also jerk) to 7500 mm/s and 5 mm/s (jerk). You have to do this in the .json files as the limits are locked within Bambu Slicer.

The result is less ringing and sharp corners:
[image]

**To sum the issues up:**
- vibration compensation is not effective or is not showing the results as it should be
- flow and pressure advance calibration is not using the printing acceleration you set in Bambu Slicer
- machine limits are locked in Bambu Slicer which kind of solve the two issues above

tapir, 2022-10-06:
Wow this is really good detective work. Will try also. +1

Argolein, 2022-10-06:
@tapir 
For you and those who want to try this: Don't forget to use the same acceleration for machine limits (.json file) and printing acceleration in Bambu Slicer. 

How I see it:
Acceleration set in machine limits is used for pressure advance calibration and the set printing acceleration in Bambu Slicer is used for the actual print. So those two values need to be identical. 

I get fairly good results since then. Ringing is not 100% gone but far better. Maybe vibration compensation will be improved in the future.

tapir, 2022-10-06:
We're talking about fdm_machine_common.json in user directory right? And where do you see this options greyed out? I couldn't ding them in UI.

SaltWei, 2022-10-11:
The machine limit is just used to show max value of Bambu printer, not the real used acceleration when printing.
The printing acceleration is in speed setting page of process preset.
We will set printing acceleration before printing calibration line, but keep the limits read-only for Bambu Lab printer.

SaltWei, 2022-10-13:
In additional, for third party printer, we will make the options editable in UI.

Patch has been merged. Please wait the next release which also include setting printing acceleration before printing calibration line

SaltWei, 2022-10-24:
Setting printing acceleration before printing calibration line，done.

SaltWei, 2022-12-26:
Already included in latest release. Close

closed 2022-12-26 by SaltWei