bambulab/BambuStudio #362: [FR] Add Wipe Inside At Start of External Perimeter

ChaosBlades, 2022-10-04:
Screenshot from SuperSlicer.
[image]
This should fix this problem. The only solution I have found due to the limitations of Studio at current is to disable all part cooling which obviously produces other issues. This is a fairly common problem in my experience specifically with PETG.
[image]
Here is the gcode from SuperSlicer with this feature enabled. The blue is the detraction.
[image]

QingZhangBambu, 2022-11-21:
I enable the setting in super slicer(2.4.58.5 edition, released on Sep 24), and print it in the Bambu X1 Carbon machine.

travel path when the wipe inside at start and end open enabled
[image]

print result in Bambu's machine, the seam is still very large
[image]

if wipe inside at the start enable, and wipe at the end off, the seam would be better but it has extra filament at the start of the perimeter, also, the connection part would be weak.
 path
[image]
print result
[image]

so, the wipe at the start will not be added to Bambu Studio, but we will try to modify the end wipe path to make the seam better. Thank you for the advice. It is helpful.

ChaosBlades, 2022-11-21:
@QingZhangBambu The point of this setting isn't to close the seam. It is to give the melted filament something to grab onto so it is not drug away from the seam. If you don't have an issue with the filament getting drug away from the steam like in my example then this setting is of no use to you.

If you want to close the gap then you would lower the Z Steam gap setting. Like I said, that isn't what this setting does.

QingZhangBambu, 2022-11-21:
sorry for the misunderstanding. I will check the z seam.

ChaosBlades, 2022-11-21:
If you want to try and reproduce this issue here is the 3MF. You would want to print this in PETG as it likes to stick to the nozzle. However my issue may be specific to my brands blend of PETG.
[1U Modular Drawer 12x6.zip](https://github.com/bambulab/BambuStudio/files/10053189/1U.Modular.Drawer.12x6.zip)

QingZhangBambu, 2023-02-06:
I use bambu's PETG. and the seam is not as bad as yours.
[image]

QingZhangBambu, 2023-02-06:
though the gap is big, but the nozzle haven't wipe extra filament to near place.

closed 2023-02-20 by QingZhangBambu

QingZhangBambu, 2023-02-20:
your issue could not be reproduced. So close the issue.