bambulab/BambuStudio #363: [Enhancement]

FreeManFreeThinker, 2022-10-04:
Since light interferes with the printers LiDAR functions (bed leveling, flow calibration), we should attempt to decrease the amount of light in the enclosure, no?

My proposal is to disable the light in the enclosure when LiDAR functions are occurring. I imagine this would help LiDAR functions overall but also those who use PEI sheets. 

I need the torch on for Timelapse and occasional inspections. I don't need it on when the LiDAR is on.

I have absolutely no idea how big of an impact the torch makes but I print in either a dark or pitch black room and think the light turning off for lidar scanning would make an improvement and seems like something very easy to implement 😊

Liamnerf, 2022-10-08:
I like this idea! Maybe then, a mod could be created to replace the stock leds with more powerful ones.

FreeManFreeThinker, 2023-08-09:
> I like this idea! Maybe then, a mod could be created to replace the stock leds with more powerful ones.

Exactly

FreeManFreeThinker, 2023-09-21:
I see lots of people on Reddit talking about how this should be a feature. I hope it's still being considered.

GuzziRaz, 2023-09-30:
+1

Also while at it, turn off lamp and extruder led when going into standby mode