bambulab/BambuStudio #366: No warning given when generated tree support overlaps

ptbsare, 2022-10-04:
[image]
[image]

ptbsare, 2022-10-04:
The tree support of individual objects are generated separately, which intersect when objects are closely orranged by the automatic placement system.  NO warnings are given.
And I think that the tree support should be generated wholly when there are multiple objects, not separately, which can avoid this problem.

tangketan, 2023-08-10:
We have warnings of all gcode paths conflicts now.
Generating tree supports of all objects conflicts with "skipping objects".

closed 2023-08-10 by tangketan