bambulab/BambuStudio #375: Feature request: Allow the user to specify the print order when printing objects sequentially

misterpeppery, 2022-10-05:
Allow the user the ability to specify the order objects get printed when printing sequentially. It is currently very difficult to know which order an object will get printed in without first slicing and checking the print order using the display sliders. Even knowing what order they would have printed in previously, small changes like a slight movement or color change can change the print order, to the point that it is difficult to specify the print order yourself. Please make this easier.

candre23, 2022-11-05:
This feature needs more traction.  The aux fan is basically useless for half the build plate if the slicer arbitrarily chooses to print on the left side first, blocking airflow to anything on the right side of the plate.

SaltWei, 2023-03-02:
> Allow the user the ability to specify the order objects get printed when printing sequentially

Already had

closed 2023-03-02 by SaltWei