bambulab/BambuStudio #376: Feature request: Bed-level calibration routine for adjusting under-bed knobs

misterpeppery, 2022-10-05:
I can see that on large prints my bed is slightly unlevel. The automatic bed leveling corrects this, but I would prefer that my bed was physically closer to level, so that auto bed leveling does not have to adjust so much and so that when watching the camera feed I do not see my bed raise and lower as far. It seems possible to me to detect the bed position at the three bed screw locations and direct the user how much to adjust each screw to more precisely level their bed. This will benefit those of us who make mostly mechanical parts because the geometry of the final print will be more accurate. This is probably a printer firmware request, since the user would need to be at the printer anyway to adjust the screws, and the calibration routine makes more sense using the touchscreen rather than Bambu Studio, but there is no github for firmware requests.

weiliang90, 2022-10-07:
I print mechanical parts and had issues with my Y and Z axis being about 0.5 degrees off from 90. It causes unacceptable tolerance issues for large parts, and the thing is: you would never know if your bed is properly trammed, unless you check your prints with a square.

I almost had to go back to my old bedslinger with BLtouch and bed visualization which made bed tramming so much easier.

rwoldberg, 2022-10-07:
I would also like this as an added feature.  While the auto bed leveling is great the closer to level you can get the bed to the x and y axis the better your prints are going to be as there is less total movement when printing.

GuzziRaz, 2023-09-30:
After performing bed levelling, the printer should obviously analyze the mesh and check whether tramming would be beneficial (extreme example: bed is perfectly flat but angled 1.5 degrees). If it is, inform the user that bed tramming is recommended and point to the instructions on the wiki (you could even include the tramming gcode in the printer for convenience).