bambulab/BambuStudio #379: Feature request:  Adjustable tree support interface density

candre23, 2022-10-06:
I'm forced to use tree supports to work around another bug, and the lack of control over those tree supports is creating a less-than-ideal situation.  In cura, you can adjust the density of the tree support interface.  Basically you can make it branch out more and support the overhang in more points, up to basically creating a solid supporting "platform".  BS un-adjustable interface density leads to widely spaced "divots" in the support interface.

[image]

Jquig76, 2022-11-06:
Would love to see the ability to adjust support density of all types. Cura lets you change that and default is 20% I believe but you can take that all the way down to 0 if you wish

clairmont32, 2022-12-26:
I just got my printer less than a week ago and discovered overall support density was a missing feature between cura and prusa slicer. +1 for this request