bambulab/BambuStudio #381: Disable Bridging.  Adjustable bridging angle.

wirlybird, 2022-10-06:
Can an option be added (check box) to disable bridging or set a minimum "unsupported area" threshold?  The slicer adds bridging where it is not needed and this can cause issues.
The slicer also adds bridging over internal infill and it counts as a top layer.  This is also pointless most of the time.  If I set 4 top layers I want 4 top layers and not 3 and a bridging layer.

We also need the ability to control the bridging angle.  Many times the bridging wants to run or span the longest part of an open area rather than span the shortest distance.  This makes for bridges that are messy or fail completely.
See red arrows and yellow arow in pictures

[image]
[image]

SaltWei, 2022-10-11:
> This makes for bridges that are messy or fail completely.

Could you share some final printing picture?

For the second picture, the area is like a ring, so you want to rotate the bridge angle for 45 degree right?

wirlybird, 2022-10-11:
Hi,

I am afraid I tossed out those prints.  I didn’t get pictures.

Basically on the second picture, the bridge with the red arrows droops down due to the long spans of opening.  So when you flip the part over and look you have strands of filament in that opening .

 

On the second picture, yes, that would be good.  The main thing is to get bridging to span the shortest distance possible.  In this situation printing the bridge at a 45 or just like the infill would help.  Being able to set or control the bridging angle, like infill, is helpful.

Ideally on this print bridging isn’t needed.  I would have turned it off and let the solid layers (purple) print over the openings.

 

Thank you for the response and I am willing and able to do any testing!

SaltWei, 2022-10-13:
> Can an option be added (check box) to disable bridging or set a minimum "unsupported area" threshold? The slicer adds bridging where it is not needed and this can cause issues.

For whether need internal bridge infill, you can turn to https://github.com/bambulab/BambuStudio/pull/252 to see more similar discussion.

For bridge angle, agree to add bridge angle setting. Will add.

wirlybird, 2022-10-13:
Thank you.  I appreciate the feedback.