bambulab/BambuStudio #385: [BUG] Windows 1.2.0.9 Filament drop down list doesn't scroll

wesleypark, 2022-10-07:
When changing the filament type in Bambu Studio, mouse wheel doesn't scroll the list. Also the side scroll doesn't appear to be clickable. You can use the keyboard to select up and down the list, but the displayed list of filaments do not change.

bambu123, 2022-10-09:
@MizuUsagi1 Could you please record a video?

MizuUsagi1, 2022-10-09:
https://user-images.githubusercontent.com/12300313/194770169-db6db9e7-8867-4181-9685-2b731b1df12b.mp4

Here you go. Trying to use mouse scroll and nothing happens here. Also, filament slot 5 doesn't even pop up with a filament window. You can get details if you click the icon on the right of the filament name.

bambu123, 2022-10-11:
It's so weird. Have you scrolled in the other direction? Is it always not working or sometimes not working?

MizuUsagi1, 2022-10-11:
It does not work all the time in either direction. Scroll works on other controls. I'm using 1920x1080p screen. Wondering if there's something with the resolution since the scroll bar looks different here than other parts of the app that does work.

RobinWhitfield, 2022-10-12:
The whole rendering of the app on your screen looks pretty horrible. Do you have any kind of scaling switched on?

bambu123, 2022-10-12:
I have no idea now. Can we talk on Discord and share your screen picture there?
My account is chunmao.guo#3806.

YifanWuBambu, 2022-12-16:
This is fixed in v1.4.

closed 2022-12-16 by YifanWuBambu