bambulab/BambuStudio #387: [BUG] Printer cannot read hot plate error

MizuUsagi1, 2022-10-07:
When using the hot plate, printer throws the cannot read plate error.

darkorb, 2022-10-07:
Did you check the [release notes](https://wiki.bambulab.com/en/x1/manual/X1-X1C-AMS-firmware-release-history) for the firmware? "Added build plate localization marker detection"

MizuUsagi1, 2022-10-07:
Yes it's this feature that's alerting. Of the 3 Bambu build plates, the cool and textured have white markings. However, the high temp plate markings are in dark brown and I think it's having trouble reading this dark marking

tangketan, 2022-10-08:
The contrast of the marker on High temp plate is just too low for accurate detection. Did you smear glue on the marker?

MizuUsagi1, 2022-10-08:
Nope. I got the high temp plate along with 2 extra high temp stickers 3 weeks ago. They're all this dark. Even the label on the side is dark.

tangketan, 2022-10-11:
OK. Then we'll need upgrade our algorithm to handle these cases. Thanks for reporting.

TheCableGuy99, 2022-12-09:
To confirm I have just started using the high-temp plate today from new and it's also reporting this for me. I questioned if I was doing something wrong in the Facebook group and everyone tells me they are having the same issue. Clearly a bug with a lot of plates (cool plate and engineering plate on flip side seems fine).
Link: https://www.facebook.com/groups/bambulabofficialusers/permalink/911088199835200

c2h5oh, 2022-12-09:
Same problem here - I think it's safe to assume it's a common one.

TheCableGuy99, 2022-12-09:
I wonder if they can supply a digital form of it that we can download and print with a label printer, cut to size and stick on? Problem solved?

liam-jones-lucout, 2023-02-20:
Still an issue for me as off Feb 2023

closed 2024-08-06 by BambulabRobot