bambulab/BambuStudio #390: [BUG / 01.02.00.09 / macOS] Bambu Studio crashes when creating a New Project

Gio76, 2022-10-07:
- **Bug Description:**
When Bambu Studio is in Full Screen, it crashes every time you click on **New Project**. If I exit the Full-Screen Mode, I can create a New Project.

- **Screenshot:**
  [image]

- **Bambu Studio Version:**
  v.01.02.00.09

- **Device Used:**
  [image]

- **Full Report Below:**
```
Process:               BambuStudio [21459]
Path:                  /Applications/BambuStudio.app/Contents/MacOS/BambuStudio
Identifier:            com.bambulab.bambu-studio
Version:               01.02.00.09 ()
Code Type:             ARM-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2022-10-07 22:17:30.6252 +0700
OS Version:            macOS 12.6 (21G115)
Report Version:        12
Anonymous UUID:        264DC695-76B6-7ABF-7B68-D135BF7DCFFC

Sleep/Wake UUID:       C9F81492-21B0-4EC0-A9D2-C145DE154BDC

Time Awake Since Boot: 89000 seconds
Time Since Wake:       1169 seconds

System Integrity Protection: enabled

Crashed Thread:        0  bambustudio_main  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGBUS)
Exception Codes:       KERN_PROTECTION_FAILURE at 0x0000006300000030
Exception Codes:       0x0000000000000002, 0x0000006300000030
Exception Note:        EXC_CORPSE_NOTIFY

Termination Reason:    Namespace SIGNAL, Code 10 Bus error: 10
Terminating Process:   exc handler [21459]

VM Region Info: 0x6300000030 is in 0x1000000000-0x7000000000;  bytes after start: 356482285616  bytes before end: 55834574799
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      commpage (reserved)         fc0000000-1000000000   [  1.0G] ---/--- SM=NUL  ...(unallocated)
--->  GPU Carveout (reserved)    1000000000-7000000000   [384.0G] ---/--- SM=NUL  ...(unallocated)
      UNUSED SPACE AT END

Thread 0 Crashed:: bambustudio_main Dispatch queue: com.apple.main-thread
0   libobjc.A.dylib               	       0x1a03c65a0 objc_msgSend + 32
1   BambuStudio                   	       0x10109b888 Slic3r::GUI::Plater::new_project(bool, bool) + 480
2   BambuStudio                   	       0x101024228 Slic3r::GUI::GUI_App::handle_web_request(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >) + 3664
3   BambuStudio                   	       0x1011d3710 Slic3r::GUI::WebViewPanel::OnScriptMessage(wxWebViewEvent&) + 948
4   BambuStudio                   	       0x102cd5c2c wxEvtHandler::SearchDynamicEventTable(wxEvent&) + 252
5   BambuStudio                   	       0x102cd58f8 wxEvtHandler::ProcessEventLocally(wxEvent&) + 88
6   BambuStudio                   	       0x102cd5720 wxEvtHandler::ProcessEvent(wxEvent&) + 56
7   BambuStudio                   	       0x102af274c wxWindowBase::TryAfter(wxEvent&) + 160
8   BambuStudio                   	       0x1029f9030 -[WebViewScriptMessageHandler userContentController:didReceiveScriptMessage:] + 876
9   WebKit                        	       0x1bef3d814 ScriptMessageHandlerDelegate::didPostMessage(WebKit::WebPageProxy&, WebKit::FrameInfoData&&, API::ContentWorld&, WebCore::SerializedScriptValue&) + 208
10  WebKit                        	       0x1bf2c0af8 WebKit::WebUserContentControllerProxy::didPostMessage(WTF::ObjectIdentifier<WebKit::WebPageProxyIdentifierType>, WebKit::FrameInfoData&&, unsigned long long, WTF::Span<unsigned char const, 18446744073709551615ul> const&, WTF::CompletionHandler<void (WTF::Span<unsigned char const, 18446744073709551615ul> const&, WTF::String const&)>&&) + 712
11  WebKit                        	       0x1bf5f45ec WebKit::WebUserContentControllerProxy::didReceiveMessage(IPC::Connection&, IPC::Decoder&) + 2112
12  WebKit                        	       0x1befe2890 IPC::MessageReceiverMap::dispatchMessage(IPC::Connection&, IPC::Decoder&) + 448
13  WebKit                        	       0x1bf204198 WebKit::WebProcessProxy::didReceiveMessage(IPC::Connection&, IPC::Decoder&) + 40
14  WebKit                        	       0x1befcee7c IPC::Connection::dispatchMessage(std::__1::unique_ptr<IPC::Decoder, std::__1::default_delete<IPC::Decoder> >) + 268
15  WebKit                        	       0x1befce78c IPC::Connection::dispatchIncomingMessages() + 468
16  JavaScriptCore                	       0x1b8bde634 WTF::RunLoop::performWork() + 200
17  JavaScriptCore                	       0x1b8bdf3d0 WTF::RunLoop::performWork(void*) + 36
18  CoreFoundation                	       0x1a060cf94 __CFRUNLOOP_IS_CALLING_OUT_TO_A_SOURCE0_PERFORM_FUNCTION__ + 28
[... 508 lines cut ...]
      "rolloutId" : "60da5e84ab0ca017dace9abf",
      "factorPackIds" : {

      },
      "deploymentId" : 240000008
    },
    {
      "rolloutId" : "61301e3a61217b3110231469",
      "factorPackIds" : {
        "SIRI_FIND_MY_CONFIGURATION_FILES" : "6216ae152a40e71046e16225"
      },
      "deploymentId" : 240000016
    }
  ],
  "experiments" : [
    {
      "treatmentId" : "5b729754-3882-4560-826f-9c0bc27d77c6",
      "experimentId" : "6192fb082171a2330e561df0",
      "deploymentId" : 400000035
    }
  ]
}
}

```

bambu123, 2022-10-09:
Thank you for reporting the issue. We will fix it soon.

YifanWuBambu, 2022-10-21:
Fixed in v.1.3.0.12.

closed 2022-10-21 by YifanWuBambu